SmingHub/Sming #405: spiffs can't find index.html in HttpServer_ConfigNetwork

icodk, 2015-11-05:
Try to use HttpServer_ConfigNetwork example : Comile OK also the filesystem as can be seen below, however when accessing with a browser I get 404 Not Found
On the terminal I get:
 path=/
Host === 192.168.4.1
parsed
Request: GET, nodata
TCP received: 400 bytes
onReadyToSendData: 1
open errno -10024

File wasn't found: index.html
send file: index.html (-1 bytes)
SEND ERROR PAGE

# Generating spiffs_bin
Checking for spiffs files
web/dev directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x40000 (262144) bytes.
Adding files in directory 'web/dev'.
Added 'bootstrap.css' to spiffs (108535 bytes).
Added 'index.html' to spiffs (12168 bytes).
Added 'jquery.js' to spiffs (86314 bytes).
Added 'settings.html' to spiffs (4096 bytes).
Added 'style.css' to spiffs (7329 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
out/firmware/spiff_rom.bin---------->0x48000

flexiti, 2015-11-06:
What SPISIZE You have and ESP Type ?

flexiti, 2015-11-06:
Also Makefile_user.mk should be modified, see reboot example.
If You don't use rboot, I don't know whether such a large file system is working properly

hreintke, 2015-12-21:
No further action needed

closed 2015-12-21 by hreintke