SmingHub/Sming #406: Problem with HttpServer Examples using Windows HTTP Clients

bondjames12, 2015-11-06:
When running any of the HttpServer examples I am unable to load "index.html" into any browser on a windows PC.
All examples compile and flash properly, as proof of this, Linux PCs and Android phones can load the webpages from the examples perfectly with no problems. When I try to load the same webpage using a windows PC the browser simply gets stuck with firefox saying "waiting for 192.168.107" in its status bar. Its the same situation for all browsers (firefox, chrome, midori) on all windows type machines. I have tried windows 10, windows 7 and windows XP with no results.

I suspect it might be some kind incompatibility with the windows VS ESP8266 TCP/IP stacks. My next step will be to setup a packet sniffer to gather more information.

I'm hoping someone else has had this problem.

As far as I know I am using the lastest SDK, etc:
Firmware Build Environment:
Windows 10 PC
Espressif-ESP8266-DevKit-v2.0.8-x86.exe
mingw with Espressif-ESP8266-DevKit-Addon.rar

SMING develop branch (as of Nov 4)

I've attached Build logs and the serial output from the HttpServer-AJAX example while trying to load index.html from a windows firefox and ubuntu firefox browsers.
Logs are here: https://gist.github.com/bondjames12/54c9b763020776c940ad

The ESP8266 shows only this when I attempt to make a http connection in windows, it reads the file but never seems to send the data over the connection.

nAccept state: 0 K=0
Free heap size=36048, K=0 
+TCP connection 
timeout updating: 70 -> 90 
TcpServer onClient: 192.168.2.112

path=/  
Host === 192.168.2.107
parsed  
Request: GET, nodata
TCP received: 290 bytes
onReadyToSendData: 1
send file: index.html (2972 bytes)
response sendHeader
response sendBody
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 1024, len: 1024
READ Template (0)
plain template text pos: 2048, len: 809

kireevco, 2015-11-06:
Interesting. Can you try curl from Windows with verbose output?

bondjames12, 2015-11-06:
Heres the output from curl ran in windows 10:
```
C:\Windows\System32>curl -V
curl 7.33.0 (x86_64-pc-win32) libcurl/7.33.0 OpenSSL/0.9.8y zlib/1.2.8 libssh2/1.4.3
Protocols: dict file ftp ftps gopher http https imap imaps pop3 pop3s rtsp scp sftp smtp smtps telnet tftp
Features: AsynchDNS GSS-Negotiate Largefile NTLM SSL SSPI libz

C:\Windows\System32>curl -v http://192.168.2.107/index.html
* About to connect() to 192.168.2.107 port 80 (#0)
*   Trying 192.168.2.107...
* Adding handle: conn: 0x21943d0
* Adding handle: send: 0
* Adding handle: recv: 0
* Curl_addHandleToPipeline: length: 1
* - Conn 0 (0x21943d0) send_pipe: 1, recv_pipe: 0
* Connected to 192.168.2.107 (192.168.2.107) port 80 (#0)
> GET /index.html HTTP/1.1
> User-Agent: curl/7.33.0
> Host: 192.168.2.107
> Accept: */*
>
```

The serial output from ESP8266:
```
onAccept state: 0 K=0                                                   
Free heap size=39408, K=0                                               
+TCP connection                                                           
timeout updating: 70 -> 90                                            
TcpServer onClient: 192.168.2.112
                                                                                               
path=/index.html                                                
Host === 192.168.2.107                                          
parsed                                                           
Request: GET, nodata                                             
TCP received: 87 bytes                                              
onReadyToSendData: 1                                              
Default server handler for: '/index.html'                        
found index.html                                                
send file: index.html (2972 bytes)                                  
response sendHeader                                                  
response sendBody                                                   
TCP connection closed by timeout: 90 (from 90)                 
TCP connection closing                                                   
~TCP connection                                                   
-TCP connection
```

bondjames12, 2015-11-06:
When I run curl from my ubuntu virtualbox (running as a guest in the same windows 10 pc) it works normally:
```
colin@colin-VirtualBox:~$ curl -v http://192.168.2.107/index.html
* Hostname was NOT found in DNS cache
*   Trying 192.168.2.107...
* Connected to 192.168.2.107 (192.168.2.107) port 80 (#0)
> GET /index.html HTTP/1.1
> User-Agent: curl/7.35.0
> Host: 192.168.2.107
> Accept: */*
> 
< HTTP/1.1 200 OK
< Connection: close
< Cache-Control: public, max-age=86400, must-revalidate
< Content-Type: text/html
< 

<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="view..............
........ cut off the rest of the html
```
ESP8266 Serial putput:
```
onAccept state: 0 K=0                                                         
Free heap size=39408, K=0                                                     
+TCP connection                                                               
timeout updating: 70 -> 90                                                    
TcpServer onClient: 192.168.2.30
                                                                                                                           
path=/index.html                                                              
Host === 192.168.2.107                                                        
parsed                                                                        
Request: GET, nodata                                                          
TCP received: 87 bytes                                                        
onReadyToSendData: 1                                                          
Default server handler for: '/index.html'                                     
found index.html                                                              
send file: index.html (2972 bytes)                                            
response sendHeader                                                           
response sendBody                                                             
TCP sent: 1460                                                                
onReadyToSendData: 2                                                          
response sendBody                                                             
Stream completed                                                              
TCP connection closing                                                       
~TCP connection                                                               
-TCP connection
```

alonewolfx2, 2015-11-06:
@bondjames12 did you tried http_confignetwork example ?
i am using on windows and working very good. but i didnt tried httpserver example

bondjames12, 2015-11-06:
I used wireshark to capture packets from a curl request from Ubuntu and from windows 10. Not sure the best way to share so I've uploaded a screen capture.

The network addresses are:
Windows 10: 192.168.2.112
Ubuntu:         192.168.2.30
ESP8266:      192.168.2.107

[image]

alonewolfx2, 2015-11-06:
[image]
here is result for chrome on windows 10. i am using custom example based on confignetwork

bondjames12, 2015-11-06:
Yes I get the same problem with the confignetwork example. Since it works for you, it must be a problem with my windows PC. Later today I'll install a fresh copy of windows 10 on real hardware and try again.

alonewolfx2, 2015-11-06:
what is your module ? what is your flash size? can you provide confignetwork compiled bin files for me ? so i can flash my module with your binary images and try

bondjames12, 2015-11-06:
My module is a ESP-12, the flash size is 4Mb
[image]

I've uploaded a binary of the HttpServer_ConfigNetwork example below. This is built in windows using the develop branch as of Nov 4, 2015. 
https://drive.google.com/file/d/0B9fkqopa9tdWaUp4eDZOTFRfWk0/view?usp=sharing

Thanks for helping me track this problem down.

alonewolfx2, 2015-11-06:
what is spiffs rom address on flash ?

bondjames12, 2015-11-06:
Sorry about that. I believe its 0x48000, I am flashing with this command:
esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x48000 out/firmware/spiff_rom.bin

bondjames12, 2015-11-07:
I believe I have narrowed my problem to my MTU setting. My windows machines use a MTU of 1400. That seems to be preventing them from communicating with the ESP8266 module. If someone could change there MTU and double check for me that would be great.

alonewolfx2, 2015-11-07:
its working for me. so propably you are right.  but i cant change mtu
settings. because of my router didnt accept mtu1400. i will try when i get
my friends home.

hreintke, 2015-11-12:
@bondjames12 : Is is OK to close this issue or do you have further information ?

bondjames12, 2015-11-13:
@hreintke I worked around the problem by changing some clients in my LAN to 1500 MTU. Problem has not been confirmed by anyone else. Even if it was its probably a limitation of code in the expressIF SDK. I would say its fine to close and maybe make a note of this possible limitation somewhere for other users.

closed 2015-12-21 by hreintke

mxaris, 2016-03-21:
i have the same problem, the web server working fine inside my LAN but when i try to access it from outside (ADSL PPPOE with MTU < 1460) it servers only the json response which are very small in size but not any other page or file. How can i reduce the MTU in http server?

bondjames12, 2016-03-21:
Hi mxaris, the problem lies with expressIFs IP stack. They use a modifier version of lwip (light weight IP). As far as I have found there is no way to change the MTU or enable IP reassembly, etc. The SDK only includes a prebuilt libraries (sdkdir/lib/liblwip.a)

There are some people that are working on replacement IP stacks for the ESP8266 based on code expressIF has released. I found some information and more links to github here: http://www.esp8266.com/viewtopic.php?f=9&t=1492

I haven't gone any further as I decided to work around the problem. 
As you can see even in this sample config (https://github.com/kadamski/esp-lwip/blob/esp8266-1.4.1/config/lwipopts.h)
These are defined which seem related.
#define TCP_MSS 1460
#define IP_FRAG                             0
#define IP_REASSEMBLY                       0

I'm just guessing but enabling these may result in a dramatic increase in the memory requirements for the IP stack. Its worth a try I guess.