SmingHub/Sming #408: m_printf causing crash when printing large (256 + ?) %s

hreintke, 2015-11-07:
When using debugf(..) with large string.c_str() esp crashes.
Did not test in extenso all possibilities/options.

@ADiea : Can you take a look at this ?

ADiea, 2015-11-07:
HI, yes it is true, there is no size checking for the %s, I have made the fix locally on my machine, also added a snprintf. 

I am way behind with the master branch(2 months maybe) and I see very much has changed, please tell me in a few words to speed up the process,: should I rebase my master and then submit a PR against master or against develop branch?

hreintke, 2015-11-07:
@ADiea : 
Thanks for the quick reaction.
Please submit a PR against Develop. I don't see a reason to have this patch in V1.4.0 as urgent.
When in develop it will go with next release.

hreintke, 2015-11-18:
@ADiea : 
Any progress on this ? 
I would like to include this into the Develop branch

hreintke, 2015-11-29:
@ADiea : 
We are probably releasing not long from now. 
Can you submit your PR ?

ADiea, 2015-12-03:
@hreintke still didn't have the time, i'll try to resync with everything today reinstall the UDK and everything. Alot has changed in the last 2 months so I hope I can get up to speed fast

ADiea, 2015-12-03:
@hreintke  Done, see above

ADiea, 2015-12-21:
@hreintke Reminder: Fix is in #461 needs integration

hreintke, 2016-03-25:
Solved

closed 2016-03-25 by hreintke