SmingHub/Sming #410: hw timer support

johndoe8967, 2015-11-08:
Hello,

I had a look in the original espressiv sources and they have an example for a hw timer.
Just to give it a try I merged it into the Basic_Blink and it works quit well with a very low jitter (approx. 1us measured against system_get_time(), a real measurement has to be done by external HW of course).

Did you ever think of supporting the hw timer? 

best regards
John

hreintke, 2015-11-08:
@johndoe8967 : 
Yes, there have been thoughts about HW Timer support and experiments of @dmarkey with various results. With implementation as separate function or as option within the Timer Class.
Maybe you can combine efforts, propose a solution and/or prepare a PR.

johndoe8967, 2015-11-08:
exactly various results...
I've a very small program compiled with Sming AND Espressif sdk
- With Espressif it works fine
- With Sming I get an exception (Fatal exception 0(IllegalInstructionCause)) after a few seconds
    -> this exception can be "cured" it I ping the ESP8266 from my laptop!!!
    -> the latency jitter is higher with Sming, so some maybe there is some interrupt???

@dmarkey: I would like to combine and exchange the findings if you like

dmarkey, 2015-11-08:
Here is mine.. seems to work fine for me...

https://github.com/dmarkey/sming_hwtimer_example

raburton, 2015-11-08:
> exactly various results...

I've had pretty reliable results, hw timers do not work on any of my devices. When I say reliable results I mean consistent and reproducible, not the good and useful kind of reliable. When used directly the function never gets called after the expected time with one timesource and with the other it gets called but the devices resets within a few seconds. Also, hardware pwm (which uses hw timers) doesn't work either.

If I could have got hw timers to work I'd have done the work to support them in sming.

johndoe8967, 2015-11-08:
@dmarkey: my example looks quite similar, I will try yours to find the differences
   I enabled the wifistation, so that I can ping my device.

@raburton: I will try the second time source, I didn't find any documentation so I didn't change it from default

johndoe8967, 2015-11-09:
@dmarkey: i checked your example and found that it behaves exactly like mine,
it is unstable and has fatal exceptions, but if I ping the device ever second it is stable over hours without problem.

@raburton: the second time source (NMI) did work for only some seconds and then stopped triggering the function, so I don't think its useful 

I still didn't find out why it is stable if I compile with the Espressif SDK..

dmarkey, 2015-11-09:
Hmm, How long would I have to wait for the instability? Also what SDK version are you using?

johndoe8967, 2015-11-10:
It takes about 10s without ping 
I used V 1.3.0 and V1.4.0, both the same behavior

raburton, 2015-11-10:
@johndoe8967 unfortunately I haven't figured out what the conflict is either, because it does appear to work ok when used without Sming. If anyone can figure this out I'd be very grateful, because I need hw pwm.

hreintke, 2015-11-10:
@dmarkey : @johndoe8967 : 
Are you using esp-open-sdk or Espressif SDK ?

dmarkey, 2015-11-10:
esp-open-sdk

johndoe8967, 2015-11-10:
the sample is working well with the Espressif SDK
If I use the esp-open-sdk it only works when pinged permanently

raburton, 2015-11-10:
>the sample is working well with the Espressif SDK
If I use the esp-open-sdk it only works when pinged permanently

That's interesting, but can you clarify what you mean by using the espressif sdk? Esp-open-sdk is just a bundle that includes the espressif sdk. What setup exactly are you using in which this works, because I'd really like to get it to work!

johndoe8967, 2015-11-10:
ok, now it gets a little bit tricky, 
the how toolchain is running on my raspberry pi 
which I used since the very beginning and now its a little bit messed up.

I'm quite sure the working installation only uses files from this site
https://github.com/esp8266/esp8266-wiki/wiki/Toolchain
But maybe I really have to buy a new flash card and set it up from scratch to verify it

hreintke, 2015-11-10:
I always had the impression that the esp-open-sdk was the "espressif sdk" in which "some components" were replaced by open source equivalents.

johndoe8967, 2015-11-10:
ok, next test
the hw timer works stable when wifi is disabled.
I also experienced a problem with GPIO interrupts and wifi
seems to be the same problem.
maybe some code in the wifi interrupt is not 
in the RAM (IRAM_ATTR)???

dmarkey, 2015-11-10:
So it's still with esp-open-sdk, does wifi affect performance while using the "native" sdk?

dmarkey, 2015-11-11:
I wonder if the Arduino SDK has the same problem

johndoe8967, 2015-11-15:
I do think that Arduino has the same problems, have a look here
https://github.com/esp8266/Arduino/issues/819

kireevco, 2015-11-15:
Maybe it's worth building a testcase and submitting to esp-open-sdk maintainer? Could be an issue with some opensource alternatives (like openlwip)

johndoe8967, 2015-11-15:
if I could tell exactly which component or class has the problem I would do it, but at the moment it could be everything, from compiler to the chip itself...

kireevco, 2015-11-15:
@johndoe8967 let's try doing something and provide @pfalcon with full test-case (what is the code, what is the platform, setup, versions, etc). What is the code, what are the effects, etc... 
In other words, just a very good description of what is going on.

johndoe8967, 2015-11-15:
ok, I had a short look at the disassembly ... and found that the DigitalWrite method is still in cached space, from documentation I know that all interrupt code has to be in IRAM.

So I tried to set the attribute IRAM_ATTR in the Sming code, still in the wrong memory.
If I do not use any cached function the timer looks stable

.text._Z12digitalWriteth
                0x0000000040236fdc       0x78 /opt/sming/Sming/compiler/lib//libsming.a(Digital.o)

johndoe8967, 2015-11-15:
Ok, I tested it over an hour and the HW Timer Blink is now stable
the same issue happens with the external GPIO interrupt, tested with an external signal generator

johndoe8967, 2015-11-15:
so how can I make a pull request? 
I'm not that familiar with github and I really tried to push my changes in digital.h to the development branch?

hreintke, 2015-11-15:
You should push your changes into a "local branch" of your sming fork
From there you can create the pull request to main repo

johndoe8967, 2015-11-15:
I've a lokal branch and I use os x SourceTree to manage my git and there is a function to generate the PR but I always get an error.

git -c diff.mnemonicprefix=false -c core.quotepath=false -c credential.helper=sourcetree push -v --tags origin refs/heads/newMaster:refs/heads/develop 
Pushing to https://github.com/anakod/Sming.git
remote: Permission to anakod/Sming.git denied to johndoe8967.
fatal: unable to access 'https://github.com/anakod/Sming.git/': The requested URL returned error: 403
Completed with errors, see above

johndoe8967, 2015-11-15:
pull request created
https://github.com/SmingHub/Sming/pull/429
issue can be closed

closed 2015-11-15 by johndoe8967