SmingHub/Sming #414: Include esptool.exe from ESP8266 Arduino project

chall3ng3r, 2015-11-09:
Please add esptool.exe from ESP8266 Arduino project which supports NodeMCU DevKit's automatic flash mode switching. Or enable this feature in Sming's version of esptool.exe.

This makes it a bit hassle free to flash without pressing the Flash + Reset buttons.

raburton, 2015-11-10:
Sming does not include binaries and does not include any version of esptool. One day there may be a  windows package with pre-compiled binaries, but it's easy enough to download that one program and stick it in your path.

closed 2015-11-10 by raburton

chall3ng3r, 2015-11-11:
This is enhancement request for future binary builds. For now I will be doing same as you suggested, adding it my user makefile.

kireevco, 2015-11-11:
I'm working on a binary build which will have a nice installer or something
that will have choices.

Sincerely,
  Dmitry Kireev
  Cell: +1 (818) 850-55-77
  Skype: kireev.co

chall3ng3r, 2015-11-11:
Looking forward to it :)

It would be awesome if this binary package supports Arduino's Boards Manager thing, that will allow Eclipse Arduino C++ plugin to be used with Sming.

tprochazka, 2015-11-17:
@kireevco Please not not force default installation path like current Chocolatey does.
Because of that I did full installation manually and it works.
For now I have just xtensa-lx106-elf on C:\ there is somewhere hardcoded path in the Sming, but I don't have time to handle it.

kireevco, 2015-11-17:
@tprochazka you can set `ChocolateyBinRoot` environment variable and choose install path this way.