SmingHub/Sming #416: Compilation problem using new ArduinoJson library

vsirvent, 2015-11-10:
Hello,

I cannot compile the new ArduinoJson library, the esptool fails. I've tested on two different computers, with ubuntu 12 and ubuntu 14 and i've got the same problem in both computers.

When compiling the app I get the following error:

AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
/bin/sh: 1: [[: not found

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000350  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002d00  3ffe8350  3ffe8350  00000430  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          000076e8  3ffeb050  3ffeb050  00003130  2**4
                  ALLOC
  3 .irom0.text   0003e142  40209000  40209000  00008840  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         0000570f  40100000  40100000  00003130  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Traceback (most recent call last):
  File "/opt/esp-open-sdk//esptool/esptool.py", line 647, in <module>
    image.add_segment(e.get_symbol_addr(start), data)
  File "/opt/esp-open-sdk//esptool/esptool.py", line 384, in get_symbol_addr
    self._fetch_symbols()
  File "/opt/esp-open-sdk//esptool/esptool.py", line 381, in _fetch_symbols
    self.symbols[fields[2]] = int(fields[0], 16)
ValueError: invalid literal for int() with base 16: 'U'
make: *** [out/build/app.out] Error 1

If I use older Sming framework and I change ArduinoJson library manually to the new one I get the same results, so I guess the problem comes with the ArduinoJson change...

Any ideas?

hreintke, 2015-11-10:
@vsirvent : 
Do you have issues with compiling Smingframework or Application ?
I am on windows -> using esptools.exe. No problems whatsoever with both.

What "older sming framework" did you use to test alternative upgrade ?

vsirvent, 2015-11-11:
Hi @hreintke,

The issue is while using esptool in the application. "older" framework I mean a sming version without the ArduinoJson updated. It seems a esptool issue...

hreintke, 2015-11-14:
@vsirvent : 
Any news on this ?
If it is an esptool issue, please close this one an submit an issue on esptool

hreintke, 2015-11-15:
@vsirvent : 
This might be an esptool issue, especially on 64 bit windows.
Other tests revealed that UDK206 has an "older" version of esptool than UDK208. 
This is confirmed by the release notes from UDK. -> In these tests, using UDK208 worked OK.

Can you test whether your "erroneous behavior" still exists when using UDK208 and report back ?

vsirvent, 2015-11-15:
Yes sure. I'll try to test it this afternoon if fallout4 gives me some time to check ;)

kireevco, 2015-11-15:
@vsirvent lol....

Just bought it on Steam.

vsirvent, 2015-11-29:
With new esptool2 compilation works fine, so I think we can close this issue.

hreintke, 2015-11-29:
Solutions are discussed/tested in  #450 #447

closed 2015-11-29 by hreintke