SmingHub/Sming #417: New Install, Cannot build

GavinAndrews, 2015-11-11:
I followed the instructions on the Windows Quick Start.

The Examples failed to import into eclipse but can import manually but does not build... same error at command line...

```
:\tools\sming\Basic_Blink>make
ake: *** c:/tools/sming/Sming: Invalid request code.  Stop.

:\tools\sming\Basic_Blink>where make
:\tools\mingw64\msys\1.0\bin\make.exe

:\tools\sming\Basic_Blink>
```

Please advise

GavinAndrews, 2015-11-11:
```
C:\tools\sming\Basic_Blink>make -version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i686-pc-msys

C:\tools\sming\Basic_Blink>
```

GavinAndrews, 2015-11-11:
Edited C:\tools\sming\Sming\Makefile-project.mk
to define (by uncommenting)
```
ESP_HOME = c:/Espressif
```
and
```
SMING_HOME = c:/tools/sming/Sming 
```

Then build SMING

```
C:\tools\sming\Sming>make
AS system/irq_check.s
CC system/flashmem.c
CC Services/FATFS/ff.c
CC Services/SpifFS/spiffs_cache.c
CC Services/SpifFS/spiffs_check.c
CC Services/SpifFS/spiffs_gc.c
CC Services/SpifFS/spiffs_hydrogen.c
CC Services/SpifFS/spiffs_nucleus.c
CC Services/SpifFS/spiffs_sming.c
CC Services/libemqtt/libemqtt.c
CC Libraries/Adafruit_GFX/glcdfont.c
CC Libraries/Adafruit_ILI9341/hspi.c
C+ system/esp_cplusplus.cpp
C+ system/m_printf.cpp
...
C+ Libraries/RF24/RF24.cpp
C+ Libraries/SDCard/SDCard.cpp
C+ Libraries/TFT_ILI9163C/TFT_ILI9163C.cpp
C+ Libraries/Ultrasonic/Ultrasonic.cpp
C+ Libraries/WS2812/WS2812.cpp
C+ Libraries/si4432/si4432.cpp
AR out/build/app_app.a
Installing libsming
Done

C:\tools\sming\Sming>
```

Then...

Edit C:\tools\sming\Basic_Blink\Makefile-user.mk t
to define (by uncommenting)
```
ESP_HOME = c:/Espressif
```
and
```
SMING_HOME = c:/tools/sming/Sming 
```
then...

```
C:\tools\sming\Basic_Blink>make
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8351|     849
    rodata|           ReadOnly Data (RAM)|    3FFE8358|    3FFE8600|     680
       bss|      Uninitialized Data (RAM)|    3FFE8600|    3FFEF0C0|   27328
      text|            Cached Code (IRAM)|    40100000|    401052F2|   21234
irom0_text|           Uncached Code (SPI)|    40209000|    4023391C|  174364
Total Used RAM : 28857
Free RAM : 53063
Free IRam : 11552
------------------------------------------------------------------------------
# Generating image...
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8351|     849
    rodata|           ReadOnly Data (RAM)|    3FFE8358|    3FFE8600|     680
       bss|      Uninitialized Data (RAM)|    3FFE8600|    3FFEF0C0|   27328
      text|          Uncached Code (IRAM)|    40100000|    401052F2|   21234
irom0_text|             Cached Code (SPI)|    40209000|    4023391C|  174364
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 28857
Free RAM : 53063
Free IRam : 11534 or 27918 if 48k IRam
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
No files found in ./files.
Creating empty out/firmware/spiff_rom.bin (16384 bytes)
out/firmware/spiff_rom.bin---------->0x34000

C:\tools\sming\Basic_Blink>
```

closed 2015-11-11 by GavinAndrews

eltonfernando, 2017-10-05:
Makefile-user.mk
SMING_HOME = c:/tools/sming/Sming
remova o espaço no final do caminho (remove the space at the end of the path)