SmingHub/Sming #418: Why Sming does not have Websocket Client????

williamesp2015, 2015-11-11:
Few months ago hrsavla added Websocket Client and added pull request but did not imported into Sming. Please add binary send and receive and add it.
https://github.com/hrsavla/Sming/tree/15549097b0681b79b1edc70e4cb27cc9b913333e/Websocket_Client

tprochazka, 2015-11-12:
I agree. It would be extremely useful.

hreintke, 2015-11-12:
There were two pull requests (#202 and #258)  which were close to being able to merge.
At that time Sming project was restructuring PR flow and  after I have seen a number of request/questions to @hrsavla to finish the  PR.

If @hrsavla or other can check functionality on Sming V1.4.0/ Develop branch and submit PR we probably can discuss and finish quickly.

azmathmoosa, 2016-02-12:
I tried and it works for me. I have Sming V2.1.0.  I had to download and place some files from his version -> PWM.cpp, PWM.h inside the SmingCore folder.  And "WebSocketClient.cpp", "WebsocketClient.h" into SmingCore/Network folder.  Then I rebuilt sming by running make inside Sming/Sming folder.  Then the example started working well.  It would be nice to see support for binary data.  Please guys, do add it.

alon24, 2016-03-15:
@azmathmoosa can u pr the changes u made to make things work?

hreintke, 2016-03-15:
@azmathmoosa : @alon24 : 
Please take also remarks from #282 into account.

azmathmoosa, 2016-03-21:
@alon24 Sorry for the delay.

The files that we are interested are "PWM.cpp" and PWM.h" inside Sming/Sming/SmingCore/.  Or just follow this link: [Here](https://github.com/hrsavla/Sming/tree/15549097b0681b79b1edc70e4cb27cc9b913333e/Sming/SmingCore) and look for those files.  Download or Copy/Paste that into a new file and place them inside your system in Sming/Sming/SmingCore.  Now go [here](https://github.com/hrsavla/Sming/tree/15549097b0681b79b1edc70e4cb27cc9b913333e/Sming/SmingCore/Network) and download "WebSocketClient.cpp", "WebsocketClient.h" and place them in Sming/Sming/SmingCore/Network folder inside your system.
Now just rebuild sming in your system.  Then you can download and run the program which is given on top of this post. (The original link).  Hope this helps.

alon24, 2016-03-21:
yes, well, i have already did a poc, and it works in rtos .
I will need to do a better check when i have the time
thanks

hreintke, 2016-05-23:
Discussionmoved to SmingRTOS

closed 2016-05-23 by hreintke