SmingHub/Sming #421: Is there a way to print out how close you are to running out of program store?

ericmnel, 2015-11-12:
The arduino environment prints out what percentage of program store flash and heap you are using, is there a facility that does the same thing in Sming?

ericmnel, 2015-11-12:
There is this information printed when you do a make all - but it gives two options about how much I have left (10330 or 26714), and doesn't seem to change at all when I compile Sming for a 4M part vs a 512k part.

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8350|     848
    rodata|           ReadOnly Data (RAM)|    3FFE8350|    3FFEA248|    7928
       bss|      Uninitialized Data (RAM)|    3FFEA248|    3FFF19D0|   30600
      text|            Cached Code (IRAM)|    40100000|    401057A6|   22438
irom0_text|           Uncached Code (SPI)|    40209000|    4024059A|  226714
Total Used RAM : 39376
Free RAM : 42544
Free IRam : 10348
------------------------------------------------------------------------------
# Generating image...
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8350|     848
    rodata|           ReadOnly Data (RAM)|    3FFE8350|    3FFEA248|    7928
       bss|      Uninitialized Data (RAM)|    3FFEA248|    3FFF19D0|   30600
      text|          Uncached Code (IRAM)|    40100000|    401057A6|   22438
irom0_text|             Cached Code (SPI)|    40209000|    4024059A|  226714
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 39376
Free RAM : 42544
Free IRam : 10330 or 26714 if 48k IRam
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x40000 (262144) bytes.
Adding files in directory 'web/build'.
Added 'bootstrap.css.gz' to spiffs (15615 bytes).
Added 'index.html' to spiffs (12466 bytes).
Added 'jquery.js.gz' to spiffs (30153 bytes).
Added 'settings.html' to spiffs (4210 bytes).
Added 'style.css' to spiffs (7710 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
out/firmware/spiff_rom.bin---------->0x44000

19:26:11 Build Finished (took 18s.12ms)

ericmnel, 2015-11-15:
The free IRam number also doesn't seem to change much based on how much code you compile into your load.

hreintke, 2015-12-21:
No further information since a long time.
Closing this issue, if question still present reopen or submit a new issue.

closed 2015-12-21 by hreintke