SmingHub/Sming #422: rBoot OTA timeouts, flashes with half received firmware

zhivko, 2015-11-12:
I can see tha downloading from htttp for the purpose of rboot is (sometimes) very slow.... If I know correctly first time I was doing otaUpdater->start it was fast.
Now whenever I triger update (over websocket command), it is very very slow.
Maybe somebody know what could be the reason ?

I use mini-httpd to serve firmware to be flashed.

[ota.txt](https://github.com/SmingHub/Sming/files/32721/ota.txt)

zhivko, 2015-11-12:
[image]
[image]
[image]

hreintke, 2015-11-12:
The cause is network/httpclient related.
Network : 
The full HTTP communication fails. The TCP transmission flow stops, esp is waiting for next package and times out. TCP connection is closed.

HTTP Client.
The HTTP Client does not use the "content length" header parameter to determine complete transmission. TCP Closure is handled as "complete completion of HTTP Body" 
HTTP Client then finishes with Code = 200 -> Complete and correct transmission 

Fixing this will require a more detailed investigation and probably adding header processing into HTTP Client

zhivko, 2015-11-12:
It is weird that this bug don't happen consistently and cannot be reproduce on a will.
Can you maybe guess/estimate what triggers error?
If you can add debug information in develop branch I can take it and try reproduce, collect log and publish it here.

hreintke, 2015-11-12:
Network errors do not consistently happen -> cannot be reproduced at will. 
No, at this moment I cannot guess/estimate the error.

I cannot reproduce here, no TCP communication failures.

Please include the line : 
`debugf("TCP ConnectionState %d",TcpClient::getConnectionState());`

In the member function of rBootHttpUpdate on the location shown below and show the serial log.

```
void rBootHttpUpdate::onTimer() {
	
	if (TcpClient::isProcessing()) return; // Will wait
	
	if (TcpClient::getConnectionState() == eTCS_Successful) {
		
		debugf("code = %d",code);
		if (!isSuccessful()) {
			updateFailed();
			return;
		}
		debugf("Firmware Loading");
		currentItem++;
		if (currentItem >= items.count()) {
			debugf("\r\nFirmware download finished!");
			for (int i = 0; i < items.count(); i++) {
				debugf(" - item: %d, addr: %X, len: %d bytes", i, items[i].targetOffset, items[i].size);
			}
			
			applyUpdate();
			return;
		}
		
	} else if (TcpClient::getConnectionState() == eTCS_Failed) {
		updateFailed();
		return;
	}
	debugf("TCP ConnectionState %d",TcpClient::getConnectionState());
	rBootHttpUpdateItem &it = items[currentItem];
	debugf("Download file:\r\n    (%d) %s -> %X", currentItem, it.url.c_str(), it.targetOffset);
	rBootWriteStatus = rboot_write_init(items[currentItem].targetOffset);
	startDownload(URL(it.url), eHCM_UserDefined, NULL);
}
```

zhivko, 2015-11-12:
[putty.txt](https://github.com/SmingHub/Sming/files/33207/putty.txt)
HEre is log with changes you proposed.
I also uninstalled mini-httpd and installed apache2 localy, but problem persist.
It looks like tcpclient stays in eTCS_Connected state for long time until it times out.

zhivko, 2015-11-16:
MAybe this issue is related to: https://github.com/SmingHub/Sming/issues/357 ?

raburton, 2015-11-17:
Quite possibly, the network stack seems a little flaky (I have my theories about the reasons for this #391).

hreintke, 2015-11-17:
They are maybe initiated both by an issue in the underlaying lwip implementation, but can certainly also be correct and caused by "not stable connectivity over wifi".
However, the #357 handling of sming is correct. If the lwip stack reports error, propagate to the application. 
This issue is #422 is not handled correctly. There is a timeout, according to the standards, sming should propagate error but doesn't. It reports OK. -> The application is not able to react correctly.

zhivko, 2015-12-21:
I can see more reliable operation with Sming development 1.5 branch, however I cannot blame certain functgionality from 1.5 dev to fix this.
I don't experience this anymore so I am OK to close this issue.

closed 2015-12-21 by zhivko