SmingHub/Sming #425: Conflicting change checked directly in to master

raburton, 2015-11-13:
I notice there was recently a push directly to master that has now created a conflict with the development branch. As it is a readme displayed on the GitHub Sming main page I can understand the desire to not put this in the development branch where it would remain unseen till the next release, however this breaks the current development model and will require manual resolution and merging at the point of the next release.

kireevco, 2015-11-13:
@raburton can you please provide link to that commit(s)?
Pushes to master should never happen any more.

raburton, 2015-11-13:
https://github.com/SmingHub/Sming/commit/d386cc1129740957eeedba8c4271b069fff761d4

raburton, 2015-11-13:
Oh, your message has changed. I guess you found it. Best option would be to commit it to develop branch and then merge that change into master. Then there will be no merge problems. That only works if there are no changes already in the affected files in develop we don't want to go into master e.g. for the readme documentation of new features in develop that aren't being merged into master yet.

kireevco, 2015-11-13:
@raburton 
> Best option would be to commit it to develop branch and then merge that change into master. Then there will be no merge problems. That only works if there are no changes already in the affected [...]

I completely agree with this standard pattern. But what if there is documentation update needed about features that are already in master, for instance changing README.md order, or fixing a typo?
Merging into master means release (since we always keep master=latest release)? So, this example with ota&rboot, what should've been a right way to add this readme change from @anakod ?

hreintke, 2015-11-14:
@raburton :  @kireevco : 
I agree with both the statements above. 
- merging to master always go first to develop 
- when master is updated we should create a release update.

Because of the implication of additional merge and release effort we should be very cautious to add changes to master.

I created a draft release 1.4.1 which I will publish when you also agree.
@kireevco : I think you should also update the links to "release" on your "download locations"

hreintke, 2015-12-08:
Releasing V2.0 shortly

closed 2015-12-08 by hreintke