SmingHub/Sming #426: SPIRead() vs spi_flash_read()

hreintke, 2015-11-13:
@raburton : 
I was investigating whether I could check if the received OTA file is an application image.
For that (proof of concept so not yet thinking on detailed implementation) I copied the check_image function from your rboot.c to rboot-api.c

When running I noticed that I got different results from reading flash (checked reading position and that was identical). 
I had to replace each **SPIRead(....)** by **spy_flash_read(...)** to get it work.
So I have an initial working version but would like to understand how it works.

Do you know the differences between the two (or where it is described) and why SPIread would give different results in root program vs application program ?

raburton, 2015-11-13:
SPIRead is a rom function, spi_read_flash is an sdk function. The sdk function is a wrapper to the rom function that unmaps memory mapped flash first and restores it afterwards. You can't use the rom flash functions while flash is memory mapped.

closed 2015-11-13 by raburton

hreintke, 2015-11-13:
Thx,
I am glad it makes sense what I did. Took some time and searching but works nicely now.
Now looking whether I can do something similar with spiffs fs.