SmingHub/Sming #427: Writing after opening a file with flag eFO_Append failing

zdrc, 2015-11-13:
On MacOS X 10.11.1 and I obtained Sming using the MacOS quickstart. The device in question is an esp201.

I get a `write errno -10021` when I try to write to a file I opened with the eFO_Append flag. Opening the file with the eFO_WriteOnly allows me to write no problem.

Any clue what's going on here?

zdrc, 2015-11-16:
How does one print the spiffs debug messages to serial? This would help me figure out what the problem is.

Are other people experiencing any problems appending text to a file?

zdrc, 2015-11-16:
It seems that fileWrite is returning 4294967295 which is the largest number that can be represented by 32 bits.

Why would SPIFFS_write be returning this?
Why does this satisfy the `res < 0` if clause that triggers the debugf error print?

zdrc, 2015-11-16:
So I discovered the problem, and I believe it needs to be fixed.

fileWrite calls SPIFFS_write and in SPIFFS_write there is a section of code,
```
if ((fd->flags & SPIFFS_WRONLY) == 0) {
  res = SPIFFS_ERR_NOT_WRITABLE;
  SPIFFS_API_CHECK_RES_UNLOCK(fs, res);
}
```

I believe this code dictates that SPIFFS_write will only work if opened when the flag is set to `eFO_WriteOnly` this becomes a problem when calling SPIFFS_write when a file is opened in `eFO_Append`

When I comment out this section it solves the problem. In reality, it should probably be changed to include situations where one is appending to a file.

I'm very new to Github, could someone else make this change?

hreintke, 2015-11-16:
This is code coming from the spiffs framework. And the location you mention is deep into the core of spiffs. If we are sure it is an issue with this framework is should be communicated to the spiffs project for solving there.
"Just removing" part of the code is definitely not the best option.

zdrc, 2015-11-16:
This is true. I'll communicate this to the spiffs project. Altering the if clause fixes the problem.

hreintke, 2015-11-17:
@zdrc : 
As this is that deep int core code and "standard functionality" probably good the check first if the problem comes from the sming filesystem layer above. F.e does it propagate the option correctly.

hreintke, 2015-11-17:
For testing you could use "native SPIFFS calls" 
There is a global parameter _filesystemStorageHandle within every sming application which holds the "spiffs *fs" needed for that : 
Usage like the below, which compiles but of course doen't run.
You need to set the appropriate parameters correctly.
```
	String fn = "testfile";

	spiffs_file test =SPIFFS_open(&_filesystemStorageHandle, fn.c_str() , 0, 0);

	char buf [21];

	SPIFFS_write( &_filesystemStorageHandle,test,buf,10);
```

raburton, 2015-11-17:
You haven't included code, but I suspect you haven't passed a file mode flag. You can't just pass `eFO_Append`, this is a modifier, you need to pass `eFO_WriteOnly` or `eFO_ReadWrite` as well.

zdrc, 2015-11-17:
@raburton Yes this is my problem. I was giving open only `eFO_Append` while I should've been giving it `eFO_Append | eFO_WriteOnly`. I misunderstood how to set the flags, and I was implementing fileOpen improperly. Unless there is a reason not to, why not change the definition from `eFO_Append = SPIFFS_APPEND` to `eFO_Append = SPIFFS_APPEND | SPIFFS_WRONLY` for clarity? Most other file I/O code that I have implemented has had separate append and write flags. Based on this I made the assumption that I wouldn't need to also pass `SPIFFS_WRONLY` along with `SPIFFS_APPEND`

Just wondering, but is there any consolidated documentation anywhere. I've just been looking at the code/examples to figure out how to use it.

raburton, 2015-11-17:
> Unless there is a reason not to, why not change the definition from eFO_Append = SPIFFS_APPEND to eFO_Append = SPIFFS_APPEND | SPIFFS_WRONLY for clarity?

Because you can also have it combined with read/write instead of just write.

> Just wondering, but is there any consolidated documentation anywhere. I've just been looking at the code/examples to figure out how to use it.

Unfortunately not, nobody is interested in writing it. Currently the samples, or asking on gitter, are the best ways to learn. Hopefully one day someone will volunteer to write some documentation.

closed 2015-11-17 by raburton