SmingHub/Sming #430: AccessPoint mode does not work more than 5 minutes

alon24, 2015-11-16:
I tried with esp sdk1.3 to run the :

HttpServer_ConfigNetwork
CommandProcessing_Debug
HttpServer_WebSocket

Examples in AP mode only:
//	WifiStation.enable(true);
//	WifiStation.config(WIFI_SSID, WIFI_PWD);
//	WifiAccessPoint.enable(false);

//	// Run our method when station was connected to AP
//	WifiStation.waitConnection(connectOk);

	WifiStation.enable(false);
	WifiAccessPoint.enable(true);
	WifiAccessPoint.config("SmartCar", "", AUTH_OPEN);
	System.onReady(connectOk);
and they do not work, it takes very long time to connect, to the server, the connection does not last very long, and websockets take a long time to reach and be processed.

This does not happen withSTA mode.
Also if STA+AP, but STA not connected (password is incorrect or ssid not found for STA), then AP mode too does not work well.

I did not check if AP works, when STA is connected

piperpilot, 2015-11-16:
I use WifiAccessPoint in my project without problem.  One thing to check is your power source.  AP mode uses a LOT of power in contrast to station mode.  If you are powering off of USB, depending on a lot of factors, you might not be giving the ESP enough power to work properly.

alon24, 2015-11-16:
I believe this is not a power issue
do you use AP mode only? or STA+AP?
Websockets? just html?
sample working program? 
which esp sdk? which sming?

alon24, 2015-11-16:
@piperpilot can you please run the CommandProcessing_Debug prog, and just click the send buttion for a minute see if it reponds (AP) - or stops responding,. and then loos connection, or get deleys

piperpilot, 2015-11-16:
Well I'm trying...just updated to develop which I haven't run in a while and getting errors with esptool.  Here's the output:

```
------------------------------------------------------------------------------
# Generating image...
Traceback (most recent call last):
  File "/opt/esp-open-sdk/esptool/esptool.py", line 615, in <module>
    image.add_segment(e.get_symbol_addr(start), data)
  File "/opt/esp-open-sdk/esptool/esptool.py", line 364, in get_symbol_addr
    self._fetch_symbols()
  File "/opt/esp-open-sdk/esptool/esptool.py", line 361, in _fetch_symbols
    self.symbols[fields[2]] = int(fields[0], 16)
ValueError: invalid literal for int() with base 16: 'U'
make: *** [out/build/app.out] Error 1
```

I'll work through tis and see if I can figure it out...never had problems with esptool before although in my own project I have been using esptool2 for a while.

alon24, 2015-11-16:
convert the project to rboot, or if on windows move to udk 208

ericmnel, 2015-11-16:
What is udk 208?

hreintke, 2015-11-16:
@ericmnel : 
UDK208 is the esp development environment / toolchain for windows.
http://www.esp8266.com/viewtopic.php?f=9&t=820
 I am not sure which version is available thru Chocolatey

alon24, 2015-11-16:
Since you asked this you probebly do not use windows 
http://www.esp8266.com/viewtopic.php?f=9&t=820

piperpilot, 2015-11-16:
@alon24 Seems to be working fine for me.  I converted it to rboot, made the modifications above that you made to turn off station and init the web server.  Been running for over 10 minutes with no problem:

```
SENT: status
Echo: status
Sming>System information : ESP8266 Sming Framework
Sming Framework Version : 1.2.0
ESP SDK version : 1.3.0
Time = 01.01.1970 00:10:08
System Start Reason : 6
```

Now one thing I do notice is over time, my machine seems to disconnect and reconnect:

```
station: 14:10:9f:da:15:fd leave, AID = 1
rm 1
add 1
aid 1
station: 14:10:9f:da:15:fd join, AID = 1
```
This happens at a pretty regular interval if there is no data being transferred.  I suspect maybe you are hitting the server during this time.

I am also on a MAC...not sure if the wifi behavior is any different than on your windows machine.

alon24, 2015-11-16:
I do see that issue
But tell me, websocket is supposed to be FAST, do you get FAST response in AP mode, to say the help command?

Or very slow compared to not STA mode.

piperpilot, 2015-11-16:
Well "FAST" is pretty subjective...I'd say it is fast.  Here is the timing using the chrome debug console:

Attached is a screenshot of the timing using the chrome debug console.  You can see that everything from receiving the command to full response is 45 hundredths of a second...I'd say thats pretty fast.

[image]

I can run the same test in station mode, but I suspect its about the same.

hreintke, 2015-11-16:
@alon24 : 
I did a similar test as @piperpilot mentioned above except : 
- I am on windows
- I did not use rboot compile option as I do not have the esptool issue

I do not have the exact timings (couldn't find (yet) similar debug console output on firefox)
but results are the same.

Reaction id fast and I do not see any difference between STA and AP mode.

piperpilot, 2015-11-16:
@alon24 
I watched the console and timed the request EXACTLY when I saw the station leave.  Here is the log...you can see there is about a 4 second delay while the station reconnects.  That is slow, but there isn't much you can do about that other than fix the station leave/join.  Not sure if that is "normal" or not.

[image]

alon24, 2015-11-16:
K so I added a keep alive on the server firing every second a websocket
request,  this seems to have fixed my issues,  but I did not have time
enough to check.

alon24, 2015-11-16:
at home,  with keep alive it's very responsive and another thing,  at my
workplace which is saturated with wifi signals my MacBook did not connect
to the local ip of esp,  I had to check with phone,.
now at home,  my MacBook does connect to the ap.  so I am guessing that had
a negative affect on my usage.
can I set the ap mode channel?

alon24, 2015-11-23:
so I solved this issue at work by:
1. setting the AP channel to different channel
2. I added a heartbeat (1s) to websocket.

This can be closed

closed 2015-11-23 by alon24