SmingHub/Sming #435: rboot -default rom size

flexiti, 2015-11-18:
@raburton 

1. uint8 updateConfig   is always true - it intentional?
2. If You by mistake first time set the wrong size of Flash (SPI_SIZE) in user application,and next set to good one the default boot conf repair itself or need to be corrected  manually?

raburton, 2015-11-18:
1. Not intentional, not sure how that happened. Thanks, fixed.
2. If you flash with wrong spi size a new config will need to be created. However, that flash size is stored in the first few bytes of flash (just before rBoot itself in the first sector), so to fix this you must rewrite the first sector (rBoot) with the correct size set. Due to a bug with most (all?) of the tools that write to the esp, or possibly it's a bug with the esp itself, when you write a sector the one after it gets erased as well. This means when you rewrite rBoot with the correct flash size the config in sector two will get erased automatically (but not deliberately) and then recreated by rBoot on the next boot.

closed 2015-11-18 by raburton

hreintke, 2015-11-19:
@raburton :  On 2 : 
That would mean that you cannot preserve the values of the rBoot config.
Should it be put a sector ahead to solve this ?

raburton, 2015-11-19:
That problem applies when rBoot itself is rewritten over serial.

hreintke, 2015-11-19:
Currently rBoot is rewritten when flashing every rBoot enabled application

raburton, 2015-11-19:
When flashing by serial using `make flash` yes, but you should only do that once unless you break something. After that you should ota, it's much much faster and the `make flash` target only flashes the first rom anyway.

commit 12b670c2 (2015-11-18): unneeded rewrite of config on boot closes #435