SmingHub/Sming #444: RC Servo Basic Sample

johndoe8967, 2015-11-22:
I've coded a basic servo sample today and committed it to my sming fork.
At the moment I didn't test with dedicated HW (will be done this week)

Is someone interested in this sample?
I could provide it as a pull request after testing

alonewolfx2, 2015-11-22:
i tried. it seems servo working 0-90 degree. is it code or bug?

johndoe8967, 2015-11-22:
oh great that you could test it :-)
it is my code and I would like to share it to all of you as part of the sming framework

robotiko, 2015-11-22:
Hi @johndoe8967,

great!

I have a  soft timer version non pushed for several months due to motion "flickering" and not clear/working HW timer.  I just started the initial port of ESP arduino Servo lib (https://github.com/esp8266/Arduino/blob/master/libraries/Servo/src/Servo.h)   in case that you want to have a look (it allows to manage up to 24 servos, 12 per timer) . 

I will try and let you know  :)

Comments: 
1) Your min and max pulse duration seem to be too restrictive  (1.000-2.000 most examples and libs move from around 544  to 2.400).
2) It would be also desirable to allow to change a single servo, not all at same time.
3) Allow to manage servo in degrees (common positioning ) instead of pulse width . (Look at write method in Arduino lib).
4) Handle a higher number of servos. ESP timer seems to be able to handle up to 12. 

Do this lib usage conflicts with ESP HW PWM? 
It seems that arduino team managed to enable two timers, so when  PWM is used, it has another enabled to drive up to 12 servos (still trying  to figure out how to drive the second HW timer  ) 

Thank you for this contribution :)

robotiko, 2015-11-22:
@alonewolfx2 @johndoe8967 
The servo limited movement might be related to the min max values 
Try to move to 544 and 2400 

There is another test to do, that is to hold a position (where Soft timer fails),  by sending the same position several times and see that motor doesn't move.

alonewolfx2, 2015-11-22:
@robotiko i tried to max 3000 but same result. seems library not working correctly

johndoe8967, 2015-11-22:
About the functionality I agree, my sample is very lean.
-> I'll have a look at it, quite sure it's not a big problem 

- min max is from a rc spec but I will increase the range
- set value for a single servo is no problem, I will add this function
- calculation of servo degree, should be possible :-)
- up to 10 servo are possible within 2ms each at 20ms refresh
     but I'm not sure if the esp8266 has enough output pins 

Yes this lib uses the HW timer, so it will conflict with the PWM.
I never tried to use the 2nd timer, but I've read somewhere that it is used in the framework itself.
I started it the sample because most HW timer samples have problems and reboot after a short time.

robotiko, 2015-11-22:
@johndoe8967 
HW specs change a little bit from servo to servo. That is the reason why the range is extended. 
About calc for servo position:
bool HardwareServo::SetValues(uint32 value[])
{
	for (uint8 i=0; i < channel_count; i++) {
		// convert degrees if value < 200
		if(value[i] < 200){
			values[i] = map(value[i], 0, 180, SERVO_MIN,  SERVO_MAX);
		}else {
			values[i] = constrain(value[i], SERVO_MIN,SERVO_MAX );
		}
	}
	calcTiming();
	return true;
}

And about the second timer.. I guess the only reference we have is the ESP arduino implementation :)

robotiko, 2015-11-22:
Tested in a MG90S servo. 
Works, but also vibrates also.

I will try with the arduino code on same config  to see if it is a timing thing.

robotiko, 2015-11-23:
Tested in esp arduino. 
It works great there with same hardware so it is a soft/ timing issue.

johndoe8967, 2015-11-23:
Thanks for the tests.
I've one question, did you enable WIFI with Arduino?
It may affect the timing an it is active in my sample

Best regards
John

robotiko, 2015-11-23:
I tried both. 
mainly with wifi off (worse case scenario).

hreintke, 2015-11-23:
As this is using HW timer. Can it be the same issue as there is with using that ? #375 

This library uses "local implementation" of hwtimer. 
Before submitting PR, please change to "sming hwtimer"

hreintke, 2015-11-23:
Sorry, I was indeed wrongly referring to HW_PWM (which is already included) instead of HW_Timer.
But still I would prefer of implementing HW_timer within sming instead of within a library.

johndoe8967, 2015-11-23:
@robotiko: Ok today I tested at my lab and found some bugs in my code
- wrong puls polarity
- 4 servo channels on one pin

I fixed it and measured the puls with my scope 
-> completely stable down to the us
Then I connected a small servo 
-> completely smooth movement, no noise

So I'm quite confident that the HW timer is working without side effects

johndoe8967, 2015-11-23:
@hreintke: Yes you are right, the HW_timer should be within the sming framework, maybe also the servo function?

robotiko, 2015-11-24:
@johndoe8967 
tested again. 
Now working very good. 
Tested with the snippet to convert the measure to degrees?
Is is CPU Freq dependent?

>So I'm quite confident that the HW timer is working without side effects
HW_PWM is not affected ?

hreintke, 2015-11-24:
Are you both on windows/espressif sdk or also on linux/mac using esp-open-sdk ?

johndoe8967, 2015-11-24:
@robotiko the degree map is not included, but I consider it :-)
first I write a HW_Timer Class which should be inside of sming, and based on that the servo function

@hreintke I use Mac / Linux raspberry with esp-open-sdk

alonewolfx2, 2015-11-24:
@robotiko what is change ? you said not working correctly and said again its working good.

robotiko, 2015-11-24:
@alonewolfx2 
There is a fix: 
read the comments from @johndoe8967 
> Ok today I tested at my lab and found some bugs in my code
>    wrong puls polarity
 >   4 servo channels on one pin
> I fixed it and measured the puls with my scope

robotiko, 2015-11-24:
@johndoe8967 
I have the fix for the map. 
If you make the PR I can merge. 
Manage in degrees and a simpler interface to manage a simple servo (something similar to arduino with attach ) would be the only improvements to consider. 
Now timing is perfect. :D

johndoe8967, 2015-11-27:
@robotiko 
I did a complete rewrite of the servo
- added Sming/SmingCore/HardwareTimer
- individual ServoChannels (with setDegree method)

=> not tested until now

robotiko, 2015-11-27:
@johndoe8967,
I had a fast look and looks great.
+-90 degrees nice :)

I won't be able to test until Sunday ( away from computer) .. But I will
test for sure.

Thanks

@robotiko <https://github.com/robotiko>
I did a complete rewrite of the servo

   - added Sming/SmingCore/HardwareTimer
   - individual ServoChannels (with setDegree method)

=> not tested until now

johndoe8967, 2015-11-28:
Great :-) 
I tested everything possible with a small unit test, but still no test with HW (scope or servo)

hreintke, 2016-01-10:
Solved by #546

closed 2016-01-10 by hreintke