SmingHub/Sming #446: How can we reduce firmware size?

alonewolfx2, 2015-11-25:
Question is simple. How can we reduce firmware sizes for small flashes (mostly sensor nodes doesnt need big flash)? I tried to remove most arduino libs but same size.

tprochazka, 2015-12-09:
I think that biggest part of the firmware is Espressif SDK
It should be possible to skip some modules, right?
I mean .a from Espressif\ESP8266_SDK\lib\
For example mesh or WPS support.

Also would be be great if is possible to write SDK part to the rom just one time and then upgrade only my part of the code. Upload would be much faster .

raburton, 2015-12-09:
> I think that biggest part of the firmware is Espressif SDK

That is correct.

> It should be possible to skip some modules, right?
I mean .a from Espressif\ESP8266_SDK\lib\
For example mesh or WPS support.

The linker leaves out all code that is not used, so there is nothing in the rom that isn't being used. This doesn't mean that it is all being used by your app though, things that Sming uses but you don't will be included and this could probably be improved.

bondjames12, 2015-12-10:
I would change the debugf macro in sming to be a blank line. Saves some save if you don't need the debug prints

alonewolfx2, 2015-12-10:
how many kb will free with this change?

bondjames12, 2015-12-10:
About 2kb?

with:    #define debugf(fmt, ...) m_printf(fmt"\r\n", ##__VA_ARGS__)

   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8354|     852
    rodata|           ReadOnly Data (RAM)|    3FFE8358|    3FFEA9DC|    9860
       bss|      Uninitialized Data (RAM)|    3FFEA9E0|    3FFF2400|   31264
      text|            Cached Code (IRAM)|    40100000|    4010586A|   22634
irom0_text|           Uncached Code (SPI)|    40202010|    4023E0BA|  245930
Total Used RAM : 41976
Free RAM : 39944
Free IRam : 10152
------------------------------------------------------------------------------
Resource            |Size(bytes)|    Used|       %Used|        Free|   %Free
--------------------|-----------|--------|------------|------------|----------
IRAM - Cached Code  |      32768|   22616|          70|       10152|      30
SPI  - Uncached Code|     253952|  245930|          96|        8022|       4
RAM  - Data         |      81920|   41976|          51|       39944|      49

without debugf : #define debugf(fmt, ...) //m_printf(fmt"\r\n", ##__VA_ARGS__)

   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8354|     852
    rodata|           ReadOnly Data (RAM)|    3FFE8358|    3FFE9CD8|    6528
       bss|      Uninitialized Data (RAM)|    3FFE9CD8|    3FFF1700|   31272
      text|            Cached Code (IRAM)|    40100000|    4010586A|   22634
irom0_text|           Uncached Code (SPI)|    40202010|    4023D72A|  243482
Total Used RAM : 38652
Free RAM : 43268
Free IRam : 10152
------------------------------------------------------------------------------
Resource            |Size(bytes)|    Used|       %Used|        Free|   %Free
--------------------|-----------|--------|------------|------------|----------
IRAM - Cached Code  |      32768|   22616|          70|       10152|      30
SPI  - Uncached Code|     253952|  243482|          95|       10470|       5
RAM  - Data         |      81920|   38652|          47|       43268|      53

closed 2016-04-20 by hreintke