SmingHub/Sming #448: GPIO16 cannot be used as output in nodemcu devkit 1.0

robotiko, 2015-11-25:
I guess it happens in all devices, but just tested on nodemcu. 
Setting pin as output, it doesn't reflect the set values.
Can  be related with the other issues to drive high number GPIO.

alonewolfx2, 2015-11-26:
Can you try deepsleep?  If it can wakeup from deepsleep than maybe it's gpio16 to rst connection related

robotiko, 2015-11-26:
Based on official bodemcu doc,
That port can be used as general gpio.
Should not be that problem.

alonewolfx2, 2015-11-26:
but if you try gpio16 to LOW, module goes to reset. how can you use it if
it connected to rst?

robotiko, 2015-12-03:
@alonewolfx2 
I set it to low and turns on the nodemcu blue led but doesn't go to reset.

alonewolfx2, 2015-12-03:
@raburton is deepsleep working correct? mean is module waking correctly
from deepsleep?

raburton, 2015-12-03:
@alonewolfx2 no idea sorry, never used deep sleep

alonewolfx2, 2015-12-04:
@robotiko it seems gpio16 working fine as output on esp-07 so propably its nodemcu board related.

hreintke, 2016-04-29:
No activity for a low period and probably not sming related

closed 2016-04-29 by hreintke