SmingHub/Sming #449: Significant http server performance fall with sdk 1.4.x

avr39-ripe, 2015-11-26:
I was wonder, why small json-files served by sming's response.sendJsonObject(stream); takes so long time to download. I make some tests
Test was made with IDENTICAL app, just recomplie sming and recompile+reflash app

1. recent sming/master + sdk 1.2.0 = 500..600 ms to download
2. sming/master + sdk 1.3.0 = 600..700 ms
3. sming/master + sdk 1.4.0 = 1.3 sec (!) 1.4 sec
4. sming/develop + sdk 1.4.1.pre5 = 1.2..1.35 sec

So tests shows that something VERY bad vas introduced with sdk 1.4.x in context of http serving

My knowledge isn't enough to troubleshoot it further, but may be somebody with more solid skills in sdk and/or sming internals can shed a light on this problem?

Can we temporary revert SDK 1.4 as mandatory dependence to sming/develop ?

hreintke, 2015-11-26:
@avr39-ripe : 
Please attach the testfiles, applcation, logfiles and results. That is a way that several tests are based on the same -> Only way to compare tests.
Is't Sming/master + sdk 1.40 a not working combination ?

alonewolfx2, 2015-11-26:
@hreintke I agree. @avr39-ripe can you add test code for more testing

avr39-ripe, 2015-11-26:
https://github.com/avr39-ripe/ValveControl/tree/pure-esp is test app.
response generated by onAJAXGetState() (https://github.com/avr39-ripe/ValveControl/blob/pure-esp/app/webserver.cpp#L82) was the test case

but same thing was with either static files or templates or json responses. greater sdk version greater time to download the same response. Moreover, response time not vary much from big, 1.5..2 kb files to several 100.300 bytes jsons it takes 1.2..1.4 sec on sming/develop + sdk 1.4 to download index.html and small json.

avr39-ripe, 2015-11-26:
For cleaner test one can pick some sming example and make test measurements with it. I think this is a good canditate for test case: https://github.com/SmingHub/Sming/blob/develop/HttpServer_AJAX/app/application.cpp#L40

My tests isn't fully complete and good, but even such tests show some trends :( bad trends.

hreintke, 2015-11-27:
@avr39-ripe : 
You might have hit a more conceptual issue within the TCP/HTTP implementation of Sming.

Can you update the following function in tcpconnection.cpp and report back.
It is probably not a final solution but should limit the impact considerably.

From : 
```
void TcpConnection::close()
{
	if (tcp == NULL) return;
	debugf("TCP connection closing");

	tcp_arg(tcp, NULL); // reset pointer to close connection on next callback
	tcp = NULL;
}
```
To 

```
void TcpConnection::close()
{
	if (tcp == NULL) return;
	debugf("TCP connection closing");

	tcp_poll(tcp, staticOnPoll, 1);
	tcp_arg(tcp, NULL); // reset pointer to close connection on next callback
	tcp = NULL;
}
```

avr39-ripe, 2015-11-28:
Hello! Your suggestion to TcpConnection::close is absolutely FANTASTIC!!! 250..350 ms on sdk1.4.x + sming/develop instead of more than 1 second!!! Thank you a lot! It must go into develop as soon as possible! Keep going with such changes in Sming NETWORKING internals!

alonewolfx2, 2015-11-29:
@avr39-ripe @hreintke approved. its very good catch and working as expected

piperpilot, 2015-11-30:
I just tested this as well and things are MUCH faster all around.  Recommend merge to develop ASAP.

alonewolfx2, 2015-12-01:
also working good on sdk 1.5

avr39-ripe, 2015-12-08:
Solved by #454 THANKS A LOT to @hreintke !!!

closed 2015-12-08 by avr39-ripe