SmingHub/Sming #452: automatic unit test?

johndoe8967, 2015-11-27:
Hello,

one question, do you think some automatic unit test would be a good idea?
I implemented some HardwareTimer, HardwareServo classes but to test all the functions an automatic  unit test would be necessary.

Did you ever think of an unit test framework so that every function could be added very easily?
best regards
John

kireevco, 2015-11-27:
@johndoe8967 that would be great. it's a part of #136 . 
I'm far from being a pro with c/c++ unit testing, but I know that there are cpptest and googletest. Googletest is newer and supposed to be better and easier to use.

Automated tests will be run by CI (currently Travis CI).

johndoe8967, 2015-11-28:
@kireevco I had a look at travis CI and googletest, both are great tools, but I'm af aid that one additional component is necessary.
Most of my test would need hardware like interrupt or timer. These tests could only be performed on the target itself or a very well implemented simulator.
Other functions might be compiled for linux, like json,... 
So I think I'm not able to implement an automatic unit test framework for all functions.

I will just set implement a small suit with will run on the target.

johndoe8967, 2015-11-28:
But it would be possible to add a Makefile for all examples.
Travis CI could build it.

kireevco, 2015-11-28:
@johndoe8967 hardware testing is a whole different beast. I'm in the process of getting jenkins cluster to work with actual hardware, still lots of stuff to do. 

How do you see the actual test? Let's say we have access to /dev/usb0, which is a nodemcu board which turns on flash mode automatically, and you have access to the terminal after it (minterm.py, for instance). Let's discuss basics, what do we assert?

johndoe8967, 2015-11-28:
Yeah I had some similar idea, work with raspberry and use it to remote flash the target through ssh commands.

First of all I would say the Sming should have a dedicated test folder.
At the moment there is no test so everything is proven by the examples. 
This would bloat the examples if it is done complete or the test is poor.

The test itself depends on the function of course.
I wrote some test cases which of course should be done with google test in my hardware servo library
https://github.com/johndoe8967/Sming/blob/master/Basic_Servo/app/application.cpp
The test runs of course only on the target.

This helps a lot to verify the functionality and can be part of a dedicated test project.
- step 1: integrate a unit test framework (e.g. googletest)
- step 2: written unit tests for basic functions which are executed manually on target
- step 3: automatically download and run tests on target

kireevco, 2015-11-28:
@johndoe8967 I think this is great. Examples right now are more for humans, and we can use them as a basis for our tests, but eventually - you're right, every library or significant piece of Sming functionality should have test/tests directory. Let's lookup what kind of directory structure projects that are using googletest is common...

We should add step 4, parse test results and assert them, and report the results back to the CI.

Raspberry pi is a good idea, cheap enough. I'm thinking of building some sort of a pool of hardware around the globe, that will be hooked up to main CI server... (some day).

johndoe8967, 2015-11-28:
sounds like a plan,
now we just need someone who starts with step 1 :-)

I tried, but I have no experience with googletest and travis.
So looking at projects already using it would be a good idea.

kireevco, 2015-11-28:
@johndoe8967 If you can make it work locally, I will help with travis.  I've never worked with googletest, but I want to explore it too:)

Well, actually, if you could start a small branch in your sming fork, we can go from there?

johndoe8967, 2015-11-28:
ok, googletest uses pthread and other stuff we don't have :-1: 
I will look for some lightweight embedded solution like:
- http://c2.com/cgi/wiki?CppUnitLite
- https://github.com/joewalnes/tinytest/

kireevco, 2015-11-28:
@johndoe8967 ah... 
maybe cppunit?

I just want to stick to something that is common amongst other developers across the board (if we can)..

johndoe8967, 2015-11-28:
@kireevco  of course, but even the simple unit test framework use something like
- fprintf (stdout, "There were %ld failures\n", failureCount);
of course we don't have stdout and also no fprintf so I have to find a way around

The CppUnitLite is aimed to be portable so I give it a try

kireevco, 2015-11-28:
@johndoe8967 (updated prev. comment)

johndoe8967, 2015-11-29:
CppUnitLite at least compiles after replacing the fprintf 
but the compiler always throws my TEST away (I think it will be "optimized") :(

kireevco, 2015-12-25:
@johndoe8967 hey, what's going on with the CppUnitLite? Could you please hel create `Tests` directory with the most basic test?

johndoe8967, 2015-12-25:
Hey, 
of course I would help to create some tests, but at the he moment the CppUnitLite is not working because of some problem with the compiler or some optimization...

The compiler dicards every test case and I don't know how to fix this.

kireevco, 2015-12-25:
@johndoe8967 can you ping me here please [image]

kireevco, 2015-12-25:
@johndoe8967 
```
~/dev/sming/Sming/CppUnitLite   Unit_Test  make
C+ Src/Failure.cpp
In file included from ../SmingCore/../SmingCore/Network/NtpClient.h:8:0,
                 from ../SmingCore/SystemClock.h:7,
                 from ../SmingCore/../Services/CommandProcessing/CommandHandler.h:14,
                 from ../SmingCore/../Services/CommandProcessing/CommandExecutor.h:13,
                 from ../SmingCore/../Services/CommandProcessing/CommandProcessingIncludes.h:14,
                 from ../SmingCore/HardwareSerial.h:14,
                 from Src/Failure.cpp:7:
../SmingCore/../SmingCore/Network/../Platform/Station.h:36:25: error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope
  SCT_EspTouch_AirKiss = SC_TYPE_ESPTOUCH_AIRKISS
```

is this what you are seeing?

(is it a right place where I built it)?

kireevco, 2015-12-25:
@johndoe8967 actually, I just upgraded to sdk 1.5.0 and here's my result of running `make` in Sming/CppUnitLite:
```
00:07:23 **** Build of configuration Default for project CppUnitLite ****
make all 
C+ Src/Failure.cpp
C+ Src/SimpleString.cpp
C+ Src/Test.cpp
C+ Src/TestRegistry.cpp
C+ Src/TestResult.cpp
AR out/build/CppUnitLite_app.a
Installing CppUnitLite
Done

00:07:24 Build Finished (took 655ms)
```

johndoe8967, 2015-12-25:
Sorry today I'm with the family, but tomorrow
Btw I wish you all a merry Xmas :)

johndoe8967, 2015-12-25:
Yes the lib compiles, also the test sample compiles, but the tests will not be executed in the device

kireevco, 2015-12-25:
Happy holidays!

How do you (supposed to) execute those tests on the device?

kireevco, 2015-12-25:
@johndoe8967 ...maybe we should evaluate outputs of some code, but evaluation should happen on the host side? like a parser? would that make sense at all?

johndoe8967, 2015-12-26:
@kireevco the output of the unit test is printed to the serial port, the framework would also count the failed tests and report it in the result.
Of course the host has to store the result ...

To run the unit tests on a desktop machine is the usual thing with libraries, the problem with sming is that is uses a lot of HW interfaces or ESP interfaces which are not available on the desktop.
For non OS libraries like this uController it would next to impossible to run most of the functions on the desktop.

hreintke, 2016-05-23:
No progress, closing for now.

closed 2016-05-23 by hreintke