SmingHub/Sming #456: HttpClient's downloadString causes exception when called a second time

jd4ever, 2015-11-30:
Hi guys,

It seems that I can't call downloadString of HttpClient from within the callback method of a previous downloadString request.

What I'd like to do is to request (get or post) some info from the server, process the info in the callback function, and do another request to the server with the same HttpClient instance. The second request however causes the ESP to crash with an exception.

To illustrate the problem, I'm using the following test code:

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#ifndef WIFI_SSID
	#define WIFI_SSID "" // Put you SSID and Password here
	#define WIFI_PWD ""
#endif

HttpClient myhttpclient;

void onSecondRequest (HttpClient& client, bool successful)
{
	if (!successful) { Serial.println("Failed"); return; }

	String response = client.getResponseString();
	Serial.println("onSecondRequest - Server response: '" + response + "'\n\r");
}

void onFirstRequest(HttpClient& client, bool successful) {
	if (!successful) { Serial.println("Failed"); return; }

	String response = client.getResponseString();
	Serial.println("onFirstRequest - Server response: '" + response + "'\n\r");

	/*********This will show that it has finished processing*************/
	Serial.println ("onFirstRequest - isProcessing: " + String(client.isProcessing() ));

	/*********The following request will cause exception***********/
	Serial.println("Sending second request");
	myhttpclient.downloadString("http://httpbin.org/ip", onSecondRequest);
}

void connectOk() {
	Serial.println("Sending first request");
	myhttpclient.downloadString("http://httpbin.org/ip", onFirstRequest);
}

void init() {
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true);

	WifiAccessPoint.enable(false);

	WifiStation.enable(true);
	WifiStation.config(WIFI_SSID, WIFI_PWD); // Put you SSID and Password here

	WifiStation.waitConnection(connectOk);
}
```

Using the test code above, the following exception is thrown:

```
onFirstRequest - Server response: '{
  "origin": "142.150.144.39"
}
'

onFirstRequest - isProcessing: 0
Sending second request
Download: http://httpbin.org:80/ip
connect to: httpbin.org
realloc 37 -> 168
TCP connection closing
-TCP connection
DNS record found: httpbin.org = 54.175.222.246
Fatal exception 28(LoadProhibitedCause):
epc1=0x4022b38d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000010, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Are there any workarounds for this? Do I have to create a new instance of HttpClient for every new request?

hreintke, 2015-11-30:
@jd4ever : 
Will have to check. Can you an additional  test for me.
Call "myhttpclient.downloadString("http://httpbin.org/ip" with an IP number instead of the httpbin.org ?

jd4ever, 2015-11-30:
Strange, when I change to the following `myhttpclient.downloadString("http://54.175.219.8/ip", onSecondRequest);` and `myhttpclient.downloadString("http://54.175.219.8/ip", onFirstRequest);`, I don't get an exception anymore, but `onSecondRequest` function is never called.

```
Sending first request
Download: http://54.175.219.8:80/ip
connect to: 54.175.219.8
TcpConnection::connect result:, 0
realloc 38 -> 169
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 40
onReadyToSendData: 2
Header pos: 210
Server === nginx
Date === Mon, 30 Nov 2015 20:19:19 GMT
Content-Type === application/json
Content-Length === 33
Connection === close
Access-Control-Allow-Origin === *
Access-Control-Allow-Credentials === true
TCP received: 247 bytes
onReadyToSendData: 1
TCP received: (null)
onFirstRequest - Server response: '{
  "origin": "142.150.144.39"
}
'

onFirstRequest - isProcessing: 0
Sending second request
Download: http://54.175.219.8:80/ip
connect to: 54.175.219.8
TcpConnection::connect result:, -15
realloc 38 -> 169
TCP connection closing
-TCP connection
pm open phy_2,type:2 0 0
```

jd4ever, 2015-11-30:
I should add that it seems that HttpClient doesn't allow me to recursively call itself. If I use two alternate HttpClient instances, then it works:

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#ifndef WIFI_SSID
    #define WIFI_SSID "" // Put you SSID and Password here
    #define WIFI_PWD ""
#endif

/*****using two alternate clients now works*******/
HttpClient myhttpclient;
HttpClient myhttpclient2;

void onFourthRequest (HttpClient& client, bool successful)
{
    if (!successful) { Serial.println("Failed"); return; }

    String response = client.getResponseString();
    Serial.println("onFourthRequest - Server response: '" + response + "'\n\r");
}

void onThirdRequest (HttpClient& client, bool successful)
{
    if (!successful) { Serial.println("Failed"); return; }

    String response = client.getResponseString();
    Serial.println("onThirdRequest - Server response: '" + response + "'\n\r");

    Serial.println("Sending fourth request");
    myhttpclient2.downloadString("http://54.175.219.8/ip", onFourthRequest);
}

void onSecondRequest (HttpClient& client, bool successful)
{
    if (!successful) { Serial.println("Failed"); return; }

    String response = client.getResponseString();
    Serial.println("onSecondRequest - Server response: '" + response + "'\n\r");

    Serial.println("Sending third request");
    myhttpclient.downloadString("http://54.175.219.8/ip", onThirdRequest);
}

void onFirstRequest(HttpClient& client, bool successful) {
    if (!successful) { Serial.println("Failed"); return; }

    String response = client.getResponseString();
    Serial.println("onFirstRequest - Server response: '" + response + "'\n\r");

    /*********This will show that it has finished processing*************/
    Serial.println ("onFirstRequest - isProcessing: " + String(client.isProcessing() ));

    /*********Using a difference instance of HttpClient will allow it to work*************/
    Serial.println("Sending second request");
    myhttpclient2.downloadString("http://54.175.219.8/ip", onSecondRequest);
}

void connectOk() {
    Serial.println("Sending first request");
    myhttpclient.downloadString("http://54.175.219.8/ip", onFirstRequest);
}

void init() {
    Serial.begin(SERIAL_BAUD_RATE);
    Serial.systemDebugOutput(true);

    WifiAccessPoint.enable(false);

    WifiStation.enable(true);
    WifiStation.config(WIFI_SSID, WIFI_PWD); // Put you SSID and Password here

    WifiStation.waitConnection(connectOk);
}
```

hreintke, 2015-11-30:
Thx,
So the original test crashed because of the DNS resolving process.
By using IP numbers you circumvent the calling of DNS. 
In the second test you get the connect result -15 : 
#define ERR_ISCONN     -15   /* Already connected.       */
But doesn't call the callback/delegate.

I've seen your workaround, so you can continue.
Is deep into the TCP Layer -> I'll check but need some time to get that done

jd4ever, 2015-11-30:
Thanks for looking into it! I'm surprised others haven't run into the same problem.

As for the crash, is it caused by the DNS resolving process if the exception happens afterwards?
```
DNS record found: httpbin.org = 54.175.222.246
Fatal exception 28(LoadProhibitedCause):
epc1=0x4022b38d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000010, depc=0x00000000
```

I thought it has something to do with calling self with a new callback function while being within a callback function?

hreintke, 2015-11-30:
Can still be, but I think the "original" crash came when httpclient/tcpclient was processing the result of the DNS, maybe/probably related to the new request in the callback. 
Also the "second issue" (error -15), can be related to that.

jd4ever, 2015-11-30:
Ah I see. Hopefully it'll get resolved soon. Thanks!

jd4ever, 2016-01-10:
Hi guys, so far I've narrowed the problem down to the following:

```
bool TcpConnection::internalTcpConnect(IPAddress addr, uint16_t port)
{
	NetUtils::FixNetworkRouting();
	err_t res = tcp_connect(tcp, addr, port, staticOnConnected);
	debugf("TcpConnection::connect result:, %d", res);
	return res == ERR_OK;
}
```

The `tcp` argument of `err_t res = tcp_connect(tcp, addr, port, staticOnConnected);` is null when downloadString is called a second time.

You can check by placing `debugf(">>>>>tcp->state: %d", tcp->state);` at the top of the internalTcpConnect function, which will cause a fatal exception at that point.

patrickjahns, 2016-03-12:
@hreintke 

I`ve been running into the same issue today  - but using `HTTPClient downloadFile`

My experience is the same as @jd4ever mentioned. When starting another request from within the callback, the ESP resets with 
```
Fatal exception 28(LoadProhibitedCause):
```

Looking at the memory dump with addr2line for the memory map:
```
xtensa-lx106-elf-addr2line.exe -e app_0.out -a 40246255
0x40246255
C:\tools\Sming\Sming/SmingCore/Network/TcpClient.cpp:219
```
Not useful since line 219 is the last line :(

jd4ever, 2016-03-14:
In case it helps, here's my workaround. It's cumbersome but it works. I made a new class that holds two HttpClient instances, and switches between the two for subsequent requests. I hacked out some parts that were only relevant to my project so it may not compile properly anymore. Hence use the code below as a guide :)

My code will also automatically re-try the http request if success is false or if it takes too long. If it fails too many times (more than 3), it'll try to reconnect to the router.

Requests.h
```
#ifndef INCLUDE_REQUESTS_H_
#define INCLUDE_REQUESTS_H_

#include <user_config.h>
#include <SmingCore/SmingCore.h>

class Requests {
protected:
	HttpClient *arrHttpClients[2]; //array stores two instances of client
	bool useSecondClient; //lets us toggle between the two instances
	HttpClientCompletedDelegate callbackFn = NULL; //holds callback function

	String urlToSend;
	HashMap<String, String> requestHeaders;
	String contentType = NULL;
	String postBody = NULL;
	int failedAttempts;
	Timer procTimer;

public:

	Requests ();

	~Requests () {
		Serial.println("Requests destructor");
	}

	void processed(HttpClient& client, bool successful);

	void callDownloadString();

	void enableWifiAndTryAgain2();
	void enableWifiAndTryAgain();
	bool downloadString (String urlArg = NULL, HttpClientCompletedDelegate callbackFnArg = NULL);

	void setRequestHeader(const String name, const String value) {
		requestHeaders[name] = value;
	}

	void setRequestContentType(String contentType) {
		this->contentType = contentType;
	}

	void clearRequestContentType(String contentType) {
		this->contentType = NULL;
	}

	void setPostBody(const String postBody) {
		this->postBody = postBody;
	}

	void clearPostBody(const String postBody) {
		this->postBody = NULL;
	}

public:
	void checkRequestTimeout ();

	bool busy;
	bool lock;
	Timer requestTimeout;
};

#endif /* INCLUDE_REQUESTS_H_ */
```

Requests.cpp
```
#include "Requests.h"

Requests::Requests () {
	// Serial.println("Requests constructor");

	arrHttpClients[0] = new HttpClient();
	arrHttpClients[1] = new HttpClient();

	useSecondClient = true;
	failedAttempts = 0;

	busy = false;
	lock = false;
}

void Requests::callDownloadString() {
	downloadString();
}

void Requests::enableWifiAndTryAgain() {
	wifi_station_connect();
	procTimer.initializeMs(5000, TimerDelegate(&Requests::callDownloadString, this)).start(false);
}

void Requests::processed(HttpClient& client, bool successful) {
	// Serial.printf(">>>Requests::processed. client %d\n", useSecondClient);
	// Serial.printf(">>>Requests::processed: %s\n", client.getResponseString().c_str());
	useSecondClient = !useSecondClient; //switch instance on next use
	busy = false;
	requestTimeout.stop();

	if (!successful) {
		failedAttempts++;
		// Serial.printf("Requests::processed - Failed! failedAttempts: %d. client %d\n", failedAttempts, !useSecondClient );
		if (failedAttempts > 2) {
			failedAttempts = 0;
			// Serial.println("Attempting reconnect");
			wifi_station_disconnect();
			procTimer.initializeMs(2000, TimerDelegate(&Requests::enableWifiAndTryAgain, this)).start(false);
			return;
		}
		procTimer.initializeMs(500, TimerDelegate(&Requests::callDownloadString, this)).start(false);
		return;
	}

	failedAttempts = 0;

	if (callbackFn) {
		callbackFn(client, successful);
	}

}

bool Requests::downloadString (String urlArg, HttpClientCompletedDelegate callbackFnArg) {
	// Serial.println(">>>Requests::downloadString called");

	if (busy || lock) {
		// Serial.println(">>>fn bsy lock. return");
		return false;
	}
	busy = true;

	if (urlArg) {
		urlToSend = urlArg;
	}

	if (callbackFnArg) {
		callbackFn = callbackFnArg;
	}
	

	//set headers
	int i;
	for (i = 0; i < requestHeaders.count(); i++) {
		// Serial.println("setting headers");
		arrHttpClients[useSecondClient]->setRequestHeader(requestHeaders.keyAt(i), requestHeaders.valueAt(i));
	}

	if (this->contentType != NULL) {
		// Serial.println("contentType is NOT null. setting");
		arrHttpClients[useSecondClient]->setRequestContentType(this->contentType);
	} else {
		// Serial.println("contentType is null");
	}

	if (this->postBody != NULL) {
		// Serial.println("postBody is NOT null. setting");
		arrHttpClients[useSecondClient]->setPostBody(this->postBody);
	} else {
		// Serial.println("postBody is null");
	}

	bool downloadStringSucceeded = arrHttpClients[useSecondClient]->downloadString(urlToSend, HttpClientCompletedDelegate(&Requests::processed, this));
	if (!downloadStringSucceeded) {
		// Serial.printf("downloadString FALSE. client %d\n", useSecondClient);
	} else {
		// Serial.printf("downloadString true. client %d\n", useSecondClient);
	}

	requestTimeout.initializeMs(45000, TimerDelegate(&Requests::checkRequestTimeout, this)).start(false);

	return true;
}

void Requests::enableWifiAndTryAgain2() {
	lock = false;
	wifi_station_connect();
	procTimer.initializeMs(20000, TimerDelegate(&Requests::callDownloadString, this)).start(false);
}

void Requests::checkRequestTimeout () {
	// Serial.println("\n\n\n\n\n\n++++++++++++\n>>>Requests::checkRequestTimeout");
	if (busy) {
		// Serial.printf(">>>Requests::checkRequestTimeout: try reconnect. client %d\n++++++++++++\n\n\n\n\n\n\n\n", useSecondClient);

		failedAttempts = 0;
		lock = true;
		arrHttpClients[useSecondClient]->JDclose();
		wifi_station_disconnect();
		procTimer.initializeMs(2000, TimerDelegate(&Requests::enableWifiAndTryAgain2, this)).start(false);
	}
}
```

Usage
```
Requests *requestsObj;

void connectOk() {
   requestsObj = new Requests();
   String postBody = "test=test";
   requestsObj->setPostBody(postBody);
   requestsObj->downloadString("YOUR SERVER",YOUR CALLBACK FN);

}
```

patrickjahns, 2016-03-14:
@jd4ever 
Thank you for the code - it`s quite useful.

For now I have also utilized a workaround - using a timer to call for subsequent requests with the same httprequest object. 

It would still be nice to track down where this is happening and resolve it - I am just a bit lost on how to track it down. My feeling is, that somehow even though it is stated that connection are closed/dropped - it seems that they are not completely closed/freed which causes the bad access

patrickjahns, 2016-03-14:
@hreintke 

I have a question related to `TcpConnection::connect(String server, int port)` and `TcpConnection::staticDnsResponse(const char *name, ip_addr_t *ipaddr, void *arg)`

In `TcpConnection::connect(String server, int port)` you connect (?)
```
return internalTcpConnect(addr, port);
```

But before in `staticDnsResponse(const char *name, ip_addr_t *ipaddr, void *arg)`  you also call internalTcpConnect 
```
dlook->con->internalTcpConnect(ip, dlook->port);
```

Any particular reason for this? It seems to me like we are issuing two connects here - I will need to investigate further once I have a running laptop again

patrickjahns, 2016-03-14:
By reading through the code and the provided logs here, I am pretty certain that the issue related to DNSLookup is caused by a race condition between `connect` and `staticDnsResponse` both calling internalTcpConnect and/or deleting the DNSLookUp object.

Some things that would be great to know: does the espressif sdk cache dns responses/hostname lookups or are they performed each time? 

Would that erroneus behaviour still occur when we lookup two different hostnames? (So the race condition would be delayed?)

A braindump of some other things:
- before issuing a new connect, it would probably be good to check the current connection status (if it is still connected or not)
- it would be a good idea to document, that reusing the http client in the callback has limitations. (But it would make certain things easier to handle - i.e. use the callback to download a list of files.)

hreintke, 2016-03-14:
@patrickjahns : 
I think the code in Tcpconnection is OK.
Take a look at http://lwip.wikia.com/wiki/DNS

The line `err_t dnslook = dns_gethostbyname(server.c_str(), &addr, staticDnsResponse, look);`

Will either 
return OK -> The IP address is known -> we can call internal connect because we now know the IP number

return ERR_INPROGESS -> not really an error but notifies that a DNS request is started.-> tcpconnect returns without internal tcpconnect, later within staticDnsResponse the internalconnect is called.

return other error -> return false, don't call internaltconnect.

hreintke, 2016-03-14:
@patrickjahns : Have looked (again) to see if I could get an idea what is wrong.
Something what needs to be investigated.
tcp_connection is not closed - tcp local var is pointing to pcb
tcp_connection starts dnsrequest -> wait for result
tcpconnection is closed -> tcp local var = NULL
staticdns request callback happens with valid IP address 
dlook->con->internalTcpConnect(ip, dlook->port); is called but con->tcp = NULL due to above closure

patrickjahns, 2016-03-14:
@hreintke 
thanks for investigating further - I will try to help as soon as I have a proper working laptop again.

A "workaround" that comes to my mind is, to always use a fresh tcpconnection for any tcpclient

hreintke, 2016-03-14:
Tcpclient is derived from tcpconnection so a new tcpclient will always be a new tcpconnection

patrickjahns, 2016-03-14:
To me it seems though, that along the chain something is keeping an
tcpclient/connection and not properly cleaning up. Otherwise it wouldn`t be
closed inbetween requests or would it?

hreintke, 2016-03-14:
agree.
tcpconnection can also be closed by lwip (reason remote client close) by calling onreceive with pbuf == NULL

patrickjahns, 2016-03-15:
@hreintke 
This assumption would confirm that we receive Error -15 when we just use the IP without dns request. 
(same as #643). 
I can`t help with testing currently (station with dev environment broke - others I can`t use for this currently) - but what would be the next steps to see where the connection is not cleaned up properly? 

I can recall that Arduino esp framework seemed to have the same issue at some point (tcp connections not properly closed/cleaned up). But I am not sure how we can confirm this assumption. 
Any thoughts/input or request I might be able to help track down this?

hreintke, 2016-03-15:
@patrickjahns : 
If the theory is correct we do have a design flaw from earliest existence of Sming.

The underlying LWIP is using one PCB (Protocol Comtrol Block) for each connection. When tcp layer closes the pcb is deleted. The Tcp connection uses this pcb (what is correct) to handle the tcp traffic.
The tcpconnection code is based on this "single pcb" -> there is one local var tcp which holds to pointer to the pcb.
But...when "reusing" the httpclient the "previous" is still not closed, and the new one is already started -> will need two pcb's.

Tcpconnection::connect : 
```
bool TcpConnection::connect(String server, int port)
{
	if (tcp == NULL)
		initialize(tcp_new());
```
the old one active is still valid -> a new one is not requested.

The design flaw is : httpclient "has one or more" tcpconnections and not httpclient  "is a" tcpconnection.

Maybe we can work around with multiple local  tcp/pcb  vars but needs detailed design.

patrickjahns, 2016-03-15:
@hreintke 

Thanks for clarifying and working out where the issue then happens. A slight remark that crossed my mind: in general any TCP Client is not a connection but has a connection. 

I don't know if it would make sense in that case, but what if we stop the inheritance of tcpconnection already at `tcpclient` and let tcpclient handle the creation/deletion of connections?

hreintke, 2016-03-15:
@patrickjahns  
Making such a change is major. There a a lot of classes which derive from /use tcpclient/connection.
"It is not the cleanest design/code" within sming. tcpclient and tcpconnection are "very close".

patrickjahns, 2016-03-16:
@hreintke 
Itself the structure is not a bad decision - each derived class is essentially a differnt form of tcp client - yet as far as I can see/tell/understand each of them just enhance the tcpclient but don`t replace it right?

Essentially my idea was:
- on construction of TCPClient the pointer to a tcp connection will be kept NULL
- as soon as we call `TcpClient::connect` it will check if tcp connection is referring to a null pointer, if not delete the previous tcp connection
- before conenct create a tcp connection

I am not sure if this would be viable - but my initial thought. If I can be of any other help with this issue let me know. IMHO believe that it would be good to resolve this design flaw (also for RTOS)

hreintke, 2016-03-16:
@patrickjahns : 
First thing that needs to be done is to verify whether the fault analysis is correct. Up till now we talking in theory and assumption.

On your solution : We have the clear or terminology. there is httpclient/tcpconnection/httpserverconnection and more. All are interrelated..And there is LWIP and the used pcb's.
On deleting the tcp connection. The pcb is the communication between lwip and sming. that has to be handled (open/close/...) correctly in order not to have trouble with lwip.

And need to be clear on the functionality we will provide. F.e will httpclient allow two or unlimited number of connection ?

On solving the design flaw. Needs to be fixed in RTOS. Whether the backport to NONOS is viable we will see when the solution design is ready.

patrickjahns, 2016-03-16:
@hreintke 
> On deleting the tcp connection. The pcb is the communication between lwip and sming. that has to be handled (open/close/...) correctly in order not to have trouble with lwip.  

With this I was not referring to changing anything lwp related - but changing the way `TcpClient` is constructed. 
For example 
```
TcpClient::TcpClient(TcpClientCompleteDelegate onCompleted, TcpClientDataDelegate onReceive /* = NULL*/)
	: TcpConnection(false), state(eTCS_Ready)
{
[....]
}
```

All of them create a `TcpConnection` object - which then creates the corresponding lwip/sming bridge

My suggestion is to move the creation of `TcpConnection` to the point when we need a connection. This way we use a clean `TcpConnection` for any connection request. And `TcpConnection` is not touched

> And need to be clear on the functionality we will provide. F.e will httpclient allow two or unlimited number of connection ?

The Problem of httpclient (and tcpclient) is not having multiple connections at once - it is allowing several subsequent connections - which (at least for a http client) is imho "correct" behavior.  

Currently the user needs to take care of instantiating a new client for a new request(or let enough time pass, until the connection is properly closed - see #643 ). A more desirable behavior would be, to have one client object that we can reuse for any http request the user would like to perform.

In my opinion - any client (http/tcp/ntp) should not have more than one open connection at a time - but allow for any number of requests

Looking at the [class hierarchy](http://sminghub.github.io/sming-api-develop/hierarchy.html) - only HttpClient/Mqtt Client are dependant on `TcpClient` - so anything server related would not change - since TcpConnection stays the same

hreintke, 2016-03-16:
@patrickjahns : 
The TcpClient does not create a TcpConnection object. It is a (derived) connection.
`class TcpClient : public TcpConnection`

As is httpclient : 
`class HttpClient: protected TcpClient`

>The Problem of httpclient (and tcpclient) is not having multiple connections at once - it is allowing several subsequent connections - which (at least for a http client) is imho "correct" behavior. 

The "original issue" is that a new connect cannot be initiated within the "oncomplete" callback. In that situation there are two active connections within a tcpclient.

Rethinking that statement : The oncompleted is now called when the data has been sent. 
Maybe we "just should postpone" the oncomplete callback to the point when the connection really is closed. (did not check what the consequences of this would be).

hreintke, 2016-03-17:
@patrickjahns : 
Please take a look at the tcpconnection branch on SmingRTOS.
I worked from the postpone idea and needed also patches on the intermediate storage in which "data to be send " is hold before connection is established. (Stream variable in tcpclient). Even in the class  MemoryDataStream

The testapplication is pasted in the basic_blink application -> can be used for testing.
As far I see the testapplication is working with the updates.

This should be regarded as a "hacked solution" -> need serious verification on (other) network activity and probably needs more changes.

hreintke, 2016-06-21:
Known issue. 
Moved to https://github.com/SmingHub/SmingRTOS/issues/128

closed 2016-06-21 by hreintke