SmingHub/Sming #466: File system getting wiped after many wake - sleep cycles

zdrc, 2015-12-03:
The SPIFFS file system seems to wipe itself after many wake - sleep cycles (~1000). As discussed on gitter the problem is believed to be that the SPIFFS garbage collection routine is erasing the first block of the SPIFFS file system, which is causing `spiffs_mount()` to reformat and wipe the file system. I've locally removed this block of code in spiffs_sming.c that believed to be at fault...
```
  if (dat == UINT32_MAX)
  {
	  debugf("First init file system");
	  spiffs_format_internal(cfg);
	  writeFirst = true;
  }
```
I'm running an ESP201 loaded with the altered code while attached to a computer to record the debug messages printed to serial. This should allow me to see if `dat`'s value is changing while the device runs.

zdrc, 2015-12-04:
By printing out the value of `dat` over serial I have verified that this is the problem. I ran the device from about midday yesterday until morning now (about 20 hours), and I recorded two instances where the value of `dat` became 0xffffffff spontaneously. The aforementioned fix worked and my file system was never wiped.

raburton, 2015-12-04:
So why is that block getting erased? Presumably that shouldn't happen. Is your filesystem still mountable and intact even after the first block as been erased (when dat == 0xffffffff)?

zdrc, 2015-12-04:
Yes. The file system is mountable and intact even after the first block has been erased. Every cycle, I'm printing the contents of a file on the file system, and it doesn't seem to affect the file.

hreintke, 2015-12-04:
@raburton : 
The SPIFFS is doing garbage collecting -> erasing the block. 
I know, from the discussion I had with Peter Andersson, that that can also be the first block.

It is a bug in Sming to assume that 0xff.. is a non valid filesystem

raburton, 2015-12-05:
@hreintke probably we'd be better calling mount and checking the return code, instead of trying to determine for ourselves if the fs is valid. Is this something your upcoming change change to enabling the magic would cover?

hreintke, 2015-12-05:
Yes.
The mount call with "Magic Off" will always return "true". In Spiffs there is no check.
I presume that is why Anakod tried to implement his code.

When "Magic On" The Spiffs mount call checks whether all blocks in the FS are "SPIFFS blocks" and return true/false depending on the outcome. It take also in account the "erased blocks due to garbage collecting" So pretty safe.

In my change I was not quite sure whether an "implicit format" is the right thing. 
Maybe better to return the mount outcome or have a parameter in the mount function stating "format when FS incorrect)

raburton, 2015-12-05:
> When "Magic On" The Spiffs mount call checks whether all blocks in the FS are "SPIFFS blocks" and return true/false depending on the outcome

Cool, lets have that!

> In my change I was not quite sure whether an "implicit format" is the right thing.
Maybe better to return the mount outcome or have a parameter in the mount function stating "format when FS incorrect)

Not a bad idea. When I added rBoot support I found that auto-format was potentially quite destructive.

pellepl, 2016-01-07:
Hey guys, need any help with this? Enabling SPIFFS_USE_MAGIC is probably the way to go as you mentioned. Do mind, that this will render previous file systems without this flag incompatible.

hreintke, 2016-01-07:
@pellepl : 
The above issue was an internal Sming problem and needed anyhow to be resolved there.
And yes, we are planning to use USE_MAGIC for better support to detect invalid filesystems.
I discussed part of it already with you in your issue #59. 

It is just on the todo and not yet done.

raburton, 2016-01-08:
@pellepl Thanks for the offer of help. I think @hreintke had this sorted, but other priorities seem to have taken over. Incompatibility could be an issue, but the sooner we do it the less users there will be and deployed devices I assume. Certainly it could go straight into the rtos port of sming, as that's still in progress there are no guarantees about compatibility.

hreintke, 2016-01-19:
@zdrc : 
I am looking into this issue to get a future proof solution for the Sming_RTOS version.
A question on the Spiffs configuration you are using : 
- Do you have a fixed spiffs area and mounting with spiffs_mount_manual(..) or spiffs_mount()
- What is the SPI_SIZE (flash size) of the esp you are using.

zdrc, 2016-01-19:
Mounting with spiffs_mount(). I have a fixed area. The flash area is 512 kB
I believe. At least that's the size of the firmware I'm flashing on to the
ESP-201. Will Sming detect the flash size automatically or does it need to
be set?

hreintke, 2016-01-20:
@zdrc : 
Thx, understand the configuration you are using.
On size of flash : that is what you have to set yourself in makefile/makefile-user.
There are defaults in makefile-project but they are not coming from the esp you are using.

I expect there are some issues in using spiffs_mount() on larger flash sizes in combination with heavy spiffs usage as you are doing. Will investigate further.

hreintke, 2016-01-27:
Will be solved in the Sming_RTOS version by using newly added MAGIC_LENGTH option from Spiffs.
This will not be backported to the Sming_NONOS version.

closed 2016-01-27 by hreintke