SmingHub/Sming #470: esptool2 path definition

tprochazka, 2015-12-05:
Makefile-project.mk now contain
```
# esptool2 path
ESPTOOL2 ?= esptool2
```
Maybe default location should be defined as to be compatible with other smin tools

```
# esptool2 path
ESPTOOL2 ?= $(SDK_TOOLS)/esptool2
```
What do you think?

alonewolfx2, 2015-12-06:
@tprochazka but esptool2 isnt a sdk tool. its came from @raburton 's repo.

tprochazka, 2015-12-06:
Yes. I know. I builded it myself and automatically put at the same place like espotool.exe and it didn't work. And I think that in the future, when new version of the Sming will be release it will be part of sdk tool distributed by Chocolatey (or any other way)

raburton, 2015-12-06:
No plans to ship esptool2 with Sming, if that changes this path could be updated. As things stand the current path is the appropriate one.

closed 2015-12-06 by raburton

tprochazka, 2015-12-07:
So why is old one esptool shipped by Sming?
Or not shipped directly but installed by Chocolatey.
I think that this is one to benefit of Sming, that it is easy to setup building environment.