SmingHub/Sming #473: SMING LED Blinking code not working

FYPStudent, 2015-12-07:
Hi kireevco, I've downloaded your sming-basic_blink eclipse project and imported it into my eclipse software work place and tried to write the program into my ESP8266 NodeMCU wifi module circuit using the eclipse. However 2 types of problems such as "Function 'digitalWrite' could not be resolved" and "Program "g++" not found in PATH " keep coming out. How do i solve this? 
Thanks alot!! 
Under the application.cpp file you provided in that project file is the code that you typed:
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define LED_PIN 2 // GPIO2 

Timer procTimer;
bool state = true;

void blink ()
{
       digitalWrite(LED_PIN, state); 
       state = ! state; 
}

void init()
{
      pinMode(LED_PIN, OUTPUT); 
      procTimer.initializeMs(1000,blink).start();
}

alonewolfx2, 2015-12-07:
@FYPStudent did you rebuild sming framework?

FYPStudent, 2015-12-08:
No. How should I do that as well?

alonewolfx2, 2015-12-08:
@FYPStudent try to follow instuctions.

```
-> git clone https://github.com/SmingHub/Sming
-> git checkout develop
-> cd sming
-> make
-> cd ..
-> cd Basic_Blink
-> make
```
 or on windows just download sming, extract c:\tools\sming, import sming and basic blink examples, use make for smingframework and than use basic blink's make target

closed 2015-12-20 by hreintke