SmingHub/Sming #474: Reset when trying to dribe GPIO9 as output in DIO Mode and nodemcu devkit 1.0

robotiko, 2015-12-07:
To not mix issues, this was initially discussed in #398 
As stated in many places, there is a limited number of ESP devices that can drive GPIO9 due to line usage for flash access in QIO mode. 
NodeMCU devkit 1.0 is mainly designed to allow two extra pins (gpio9 and gpio10)  and amica devices are being reported as working ones (I'm testing on two amica devkits 1.0).
GPIO10 works without issue, but after setting gpio9 as output, trying to drive it high or low, device resets. 
There are a lot of ESP gpio9 issues posted in different projects (as reference https://github.com/nodemcu/nodemcu-firmware/issues/809 ) but all seem to work on devkit 1.0  once device is flashed in DIO mode.
It also seems that after flashing, it wont start flashing until device is reset.  
I tried as rboot and not rboot  dio flashing .. with no luck.  We recently identified that DIO flash was not being propagated to flashing tools and after fixing, the result was the same. 
Tested in Windows/eclipse with SDK 1.4 and Sming dev version. 
Any idea?  working configs
Thanks

flexiti, 2015-12-07:
Normally four pins of ESP8266 are connected to the Flash memory (QIO). I do not know whether the  two-bit flash memory (DIO) should not have these two pins cut off from flash (gpio9 and gpio10). Viewed nodemcu ESP8266MOD scheme?
They use not ESP-12 but ESP8266MOD
btw. In this mode (DIO), the flash memory is probably slower two times

flexiti, 2015-12-07:
[image]

flexiti, 2015-12-07:
[image]

robotiko, 2015-12-07:
@flexiti this would also mean that they should be available even in not DIO mode (becuase it is always dio mode) and should fail in QIO mode and be always slower than any esp12?

Anyway,still gpio9 resets the device.

flexiti, 2015-12-07:
@robotiko 
This means that if you want to use the port 9 and 10 ,you need to have the appropriate ESP hardware.
If You don't have, you must connect flash pins (3 - WP and 7 - HOLD) to VCC and dissconect it from ESP8266 chip.
Then you can use these GPIOs 9 and 10,
When flashing firmware you must select DIO mode.
In this case, these two ports are not used to flash, and can be accessed as an additional ports.

DIO mode use 2 bits serial, QIO 4 bits, therefore QIO is faster.

alon24, 2015-12-07:
@flexiti
Are you saying that you need to physically disconnect the pins ?

flexiti, 2015-12-07:
@alon24
this 2 pins from esp chip should not be connected to the Flash memory, see diagram above.
Here is good explanation
http://smarpl.com/content/esp8266-esp-201-module-freeing-gpio9-and-gpio10
another module, but that's about it.

robotiko, 2015-12-07:
ok @flexiti 
The thing is that I'M USING NODEMCU DEVKIT 1.0, from amica (V2 board) 
Here GPIO9 and GPIO10 should work and be accessible.
I'm using GPIO10 without issues (based in what your said, this can only be possible if not wired).
Since there are not triwired FLASH, if GPIO10 works, it is a Biwired Flash GPIO9 should work too (as manufacturer states).
[image]
[image]

I'm starting to think after looking at http://esp8266.ru/esp8266-pin-register-strapping/  that it can be a booting config!

flexiti, 2015-12-07:
If You look on schematic You can see that in DIO mode port GPIO10 is conected to Flash WP, and GPIO9 to Flash HOLD.
This could easily explain why accidentally GPIO10 work and GPIO9 not.
I would rather was not based on assumptions, only checked how you have connected a flash.

slaff, 2016-12-06:
@robotiko Feel free to reopen the ticket is you have more information related to this issue.

closed 2016-12-06 by slaff