SmingHub/Sming #476: wifi_station_set_config saves to flash

icodk, 2015-12-08:
Calling frequently  WifiStation.config() will also call `wifi_station_set_config()` (in the `WifiStation` class) , which saves the configuration to flash. Writing frequently to flash will eventually destroy the flash locations written to. Why not use `wifi_station_set_config_current()` which does not write to flash?
Writing configuration to flash is probably a result of early requirement from the AT firmware.

raburton, 2015-12-09:
Do we call this frequently? It doesn't look like we do, so it's up to the user app how often this gets called. If we didn't use this we wouldn't save the connection info, and most of the time that's what we want to do. However, having the option of not doing would be nice and could easily be achieved by adding an extra boolean parameter with suitable default (for backwards compatibility).

tprochazka, 2015-12-09:
I think that is not necessary to save connection config at all, because Expressif SDK do it automatically. 

I just call WifiStation.enable(true); and it automatically connect to the last used AP, also after power lost.

Is also possible to use ```if (WifiStation.getSSID().length() > 0) {``` to recognize if some AP is stored in the permanent memory.

So I'm using ```WifiStation.config(network, password);``` just if I want to connect to different AP or ```WifiStation.config("", "");``` if I want to clear the connection memory.

raburton, 2015-12-09:
> I think that is not necessary to save connection config at all, because Expressif SDK do it automatically.

@tprochazka you have missed the point, we are talking about the sdk saving the details.

tprochazka, 2015-12-09:
Oh. Right. But it doesn't changing my note, that example HttpServer_ConfigNetwork is bad because it call WifiStation.config on every program start and also store the same data separately to the SPIFFY partition.

But I agee that should be some way how to connect to AP without store settings to the flash. For example if you have device which will automatically connect to all found APs and looking for best one.

hreintke, 2015-12-21:
@tprochazka : 
Is the discussion finalized ? 
if there is the need for updating the HttpServer_ConfigNetwork example, can you provide a PR for that ?

icodk, 2015-12-21:
I can see that there are different meanings. raburton suggest that saving will be optional. This will makes people aware that there can be an issue in a long running applications. I will vote for this option, what others say ?

icodk, 2016-01-01:
Closed on no activity/interest

closed 2016-01-01 by icodk

raburton, 2016-01-01:
I was still interested, but it needs a bit more south than I first thought and night need to change the api slightly. Just not had time yet.