SmingHub/Sming #477: Huge program can't boot

icodk, 2015-12-08:
My (probably) huge sming based program can't boot any longer.
I was adding code (and const data) to the program and all worked fine(build download and run)
However after adding some new code today, still compiles and downloads without a problem but do not get beyond the boot message. If I remove this string or comment out some code then everything is fine Details:
sming V1.4.0 on  Windows, installed with Chocolatey.
Boot message(more details follows):
     ets Jan  8 2013,rst cause:2, boot mode:(3,6)
    
    load 0x40100000, len 22364, room 16
    tail 12
    chksum 0x7d
    ho 0 tail 12 room 4
    load 0x3ffe8000, len 880, room 12
    tail 4
    chksum 0x73
    load 0x3ffe8370, len 13652, room 4
    tail 0
    chksum 0xc6
    csum 0xc6
    csum err
    ets_main.c

My link script is the standard sming 'eagle.app.v6.cpp.ld' with a modified line:
`irom0_0_seg :                         org = 0x40209000, len = 0xF0000`

The download command:
`/c/Espressif/utils/esptool.exe -p COM2 -b 256000 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x54000 out/firmware/spiff_rom.bin`

Compiler/linker output:
`#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8370|     880
    rodata|           ReadOnly Data (RAM)|    3FFE8370|    3FFEB8C4|   13652
       bss|      Uninitialized Data (RAM)|    3FFEB8C8|    3FFF32E8|   31264
      text|            Cached Code (IRAM)|    40100000|    4010575A|   22362
irom0_text|           Uncached Code (SPI)|    40209000|    402518C6|  297158
Total Used RAM : 45796
Free RAM : 36124
Free IRam : 10424
------------------------------------------------------------------------------
\# Generating image...
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8370|     880
    rodata|           ReadOnly Data (RAM)|    3FFE8370|    3FFEB8C4|   13652
       bss|      Uninitialized Data (RAM)|    3FFEB8C8|    3FFF32E8|   31264
      text|          Uncached Code (IRAM)|    40100000|    4010575A|   22362
irom0_text|             Cached Code (SPI)|    40209000|    402518C6|  297158
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 45796
Free RAM : 36124
Free IRam : 10406 or 26790 if 48k IRam
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x40000 (262144) bytes.
Adding files in directory 'web/build'.
Added 'bootstrap.css.gz' to spiffs (15615 bytes).
Added 'index.html' to spiffs (12255 bytes).
Added 'ipsettings.html' to spiffs (4208 bytes).
Added 'jquery.js.gz' to spiffs (30153 bytes).
Added 'lockcfg.html' to spiffs (4025 bytes).
Added 'style.css' to spiffs (7329 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
out/firmware/spiff_rom.bin---------->0x54000`

I don't need  OTA/FOTA. Just want the maximum space available (nodemcu with ESP-12 and a W25Q32BV =>4MBYTE)
Any help will be appreciated. Need more info? let me know

raburton, 2015-12-09:
>chksum 0xc6
csum 0xc6
csum err
ets_main.c

Suggests there is a problem with the checksum on the iram sections, but the two checksum values given match. I know there have been reports of large apps failing before pre-rBoot, but I don't know any more details about that. I suggest giving rBoot a go and seeing if that helps, you don't need to use OTA, but it will get you through the rom loader and into a software one where if might just work better or give you more info if there is a problem.

icodk, 2015-12-09:
Thanks, I will give it a try (rBoot). I could use some instructions for one big rom. The rBoot doc talks only (as far as I understand) about generating two roms. Obviously one big rom can be bigger then each of the two roms. 
My be my program is too big. How can I see how much space I have left ? In the ESP/Arduino environment you get an indication of program space usage. Where can I see it in Sming ?

raburton, 2015-12-09:
Just use the first only (written to 0x02000) and discard the second one, if you don't want OTA support.

Picatek, 2015-12-09:
Will do that later today and let you know
Thanks

Best Regards
Itamar Cohen
Picatek
Ahornvej 27
DK-7700 Thisted
Denmark
www.picatek.dk
it@picatek.dk
M:+45 29 89 07 82

icodk, 2015-12-09:
I noticed the following  flashing the big program that does not boot
In the Not working (bigger program) the second flashing command actually override the first:
The first ends somewhare after 0x00009000 (at 0x00009000 +37888=>0x0x00009400) and the second starts at 0x00009000. So the last 1024 bytes from the first flashing are overwrited. And I am using the 'standard' sming link script with the folowing modification:
    irom0_0_seg :                         org = 0x40209000, len = 0xF0000
and make files. 
In the working flashing (after i commented out some code) there is no overlapping.
Any idea how to correct it ?

    ************ NOT WORKING
    erase size : 20480
    0x00000000 -> 0x00009000    37888	bytes

    erase size :  270336
    0x00009000 ->  0x00051800  Written 297984 bytes
    
    erase size :  475136
    0x00054000 -> 0x000d3c00  Written 524288 bytes 

    ****************** WORKING
    erase size : 20480
    0x00000000 ->0x00008c00 Written 36864

    erase size :  270336
    0x00009000 ->	0x00051800 Written 297984 

    erase size :  475136

    0x00054000 -> 0x000d3c00 Written 524288

hreintke, 2015-12-10:
@icodk : 
You can also change the "starting address" of the irom0_0_seg and update the org = 0x40209000 to a value that will fit your code of the previous area.

But then you also need to update the espool flash location. F.e the line below : which one in make-file-project.mk depends on your options on SPIFFS .

```
$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin $(SPIFF_START_OFFSET) $(SPIFF_BIN_OUT)
```
In `0x09000 $(FW_BASE)/0x09000.bin` the first  `0x09000` should be the same as in your link file.
I think the second one stays at the 0x09000, as it is "just the filename that is created"
`

icodk, 2015-12-10:
Looks like 0x40209000 is a magic number I tried 0x40209400,or 0x40209800. Everything compiles OK and also the 0x00000 is downloaded OK but then I get:
Exception: Failed to enter Flash download mode for the second part. 
Resetting the chip I get endless  boot messages: 
 `Fatal exception (28):`
  `epc1=0x40202764, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000402, depc=0x00000000`

By the way the starting address also changes the bin file name accordingly and you have to edit the Makefile-project.mk to reflect the change, also the spiff size calculations:
`SPIFF_START_OFFSET = $(shell printf '0x%X\n' $$(( ($$($(GET_FILESIZE) $(FW_BASE)/0x02010.bin) + 16384 + 36864) & (0xFFFFC000) )) )`

raburton, 2015-12-10:
>Looks like 0x40209000 is a magic number

There is nothing magic about 0x9000, the split is arbitrary and you can use any value.

> Exception: Failed to enter Flash download mode for the second part. 

You didn't try to flash the rom to 0x4020nnnn did you?, just flash it to 0xnnnn.

icodk, 2015-12-10:
This is the command line from the Makefile-project.mk
```$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x9400 $(FW_BASE)/0x09400.bin $(SPIFF_START_OFFSET) $(SPIFF_BIN_OUT)```
and the line in the ld script is:
` irom0_0_seg :                         org = 0x40209400, len = 0xF0000`
No mater which start address I choose, if it is not 0x9000 (ex. 0x9400) it will not download the bin file

hreintke, 2015-12-10:
Just to be sure.
There are multiple "flash statements" in the makefile used for different options, 
You updated the one you are using ?

icodk, 2015-12-10:
As far as I can see in the console:
`/c/Espressif/utils/esptool.exe -p COM2 -b 256000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0x9400 out/firmware/0x09400.bin    0x54000 out/firmware/spiff_rom.bin`

raburton, 2015-12-10:
Now that I think about it, doesn't the flasher require you to write from a sector boundary, sine it can only erase by sector? Try using 0xA000 instead, as that's a multiple of 4k.

hreintke, 2015-12-10:
@raburton : 
Same thoughts.
I was testing just this 0x9400 -> failed, 0xA000 -> works

icodk, 2015-12-10:
This did the trick, Now my huge program works and can be even bigger:-)
Many thanks for first class support and engagement.

closed 2015-12-10 by icodk