SmingHub/Sming #479: Proper use ICACHE_FLASH_ATTR

flexiti, 2015-12-10:
Can we use this attribute correctly?
At first glance, it looks exactly the opposite.
For a function that we want to act quickly and stable we set ICACHE_FLASH_ATTR ,
and it means that this snippet code is located run from the Flash (the slower).

Or it is somehow reversed in Sming ?

flexiti, 2015-12-11:
From espressif:
>code with ICACHE_FLASH_ATTR will end up in irom (flash) only.
code without ICACHE_FLASH_ATTR will end up in irom (flash) and will be copied to iram (ram) on bootup.

>The iram space is quite limited, including space needed by SDK and other libraries you will only have a few kbytes for your own application.**So add the phrase ICACHE_FLASH_ATTR to every function you declare**.

hreintke, 2015-12-11:
From : Comment Anakod on #182 

>ICACHE_FLASH_ATTR isn't needed, because Sming building system locate all code to IROM by default. Instead you should mark with iram attrinbute to methods what should go to IRAM memory.

I don't know exactly where that is defined but seeing the result of a Sming build I must assume that the statement is OK.

flexiti, 2015-12-13:
OK, we have all in a flash as default, and only *IRAM_ATTR* marked run from IRAM.
I tested, program from IRAM is about 15% faster.

closed 2015-12-13 by flexiti