SmingHub/Sming #480: rtos lwip link differences

hreintke, 2015-12-11:
@raburton @All : Maybe someone can give a direction to solve.

I have a repo in which I have the experimental Sming/RTOS SDK implementation.
When running the Sming which is build from there, the network/lwip environment does not work as expected.

So, I created a "minimal application" which mimics the basic implementation of the lwip usage in sming.
From there I get results I do not understand.

1/ I do not change any of the files except makefile 
2/ When in makefile : `MODULES	= user appinit`
The application works as expected and properly echo's the received messages.
3/ When in makefile : `MODULES	= user system appinit SmingCore RTOS_EC Wiring SmingCore\Platform SmingCore\Network $(filter %/, $(wildcard Services/*/))`
The application does not work as the tcp_buf() always returns zero (available space for tcp messages to be sent) 
4/ The mapfiles show minimal differences, in .debug_info, .debug_abbrev, .debug_line
I can't be sure there but does that not seem to explain the different behavior.
5/ All tests : Same environment : ESP, Compiler, Network, esptool, application,.....

[firmware_all.txt](https://github.com/SmingHub/Sming/files/59452/firmware_all.txt)
[firmware_user.txt](https://github.com/SmingHub/Sming/files/59451/firmware_user.txt)
[user_main_example.txt](https://github.com/SmingHub/Sming/files/59453/user_main_example.txt)
[Makefile.txt](https://github.com/SmingHub/Sming/files/59475/Makefile.txt)
.

alonewolfx2, 2015-12-11:
@hreintke can you share full project ?

hreintke, 2015-12-11:
Just pushed to Github.

Beware : Not complete !! , not working !!

https://github.com/hreintke/Sming_RTOS.git

alonewolfx2, 2015-12-11:
is it compiling on windows ?

hreintke, 2015-12-11:
Yes, UDK 209

alonewolfx2, 2015-12-11:
ups i cant build

```
D:\yedek\mb\Sming_RTOS\ESP_RTOS_POC>make
CC system/flashmem.c
In file included from include/user_config.h:4:0,
                 from system/flashmem.h:15,
                 from system/flashmem.c:1:
include/esp_systemapi.h:39:31: error: unknown type name 'ETSTimer'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds,
bool repeat_flag, int isMstimer);
                               ^
include/esp_systemapi.h:40:30: error: unknown type name 'ETSTimer'
 extern void ets_timer_disarm(ETSTimer *a);
                              ^
include/esp_systemapi.h:41:29: error: unknown type name 'ETSTimer'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void
*parg);
                             ^
include/esp_systemapi.h:41:42: error: unknown type name 'ETSTimerFunc'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void
*parg);
                                          ^
system/flashmem.c: In function 'flashmem_erase_sector':
system/flashmem.c:100:3: error: implicit declaration of function
'WRITE_PERI_REG' [-Werror=implicit-function-declaration]
   WRITE_PERI_REG(0x60000914, 0x73);
   ^
cc1.exe: all warnings being treated as errors
make: *** [build/system/flashmem.o] Error 1
```

hreintke, 2015-12-11:
pushed a additional directory with files from sdk which needs updating

alonewolfx2, 2015-12-11:
still same error. i cloned again,i cloned esp rtos sdk again but same result.

```cpp
D:\yedek\mb\Sming_RTOS\ESP_RTOS_POC>make
CC system/flashmem.c
In file included from include/user_config.h:4:0,
                 from system/flashmem.h:15,
                 from system/flashmem.c:1:
include/esp_systemapi.h:39:31: error: unknown type name 'ETSTimer'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
                               ^
include/esp_systemapi.h:40:30: error: unknown type name 'ETSTimer'
 extern void ets_timer_disarm(ETSTimer *a);
                              ^
include/esp_systemapi.h:41:29: error: unknown type name 'ETSTimer'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void *parg);
                             ^
include/esp_systemapi.h:41:42: error: unknown type name 'ETSTimerFunc'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void *parg);
                                          ^
system/flashmem.c: In function 'flashmem_erase_sector':
system/flashmem.c:100:3: error: implicit declaration of function 'WRITE_PERI_REG' [-Werror=implicit-function-declaration]
   WRITE_PERI_REG(0x60000914, 0x73);
   ^
cc1.exe: all warnings being treated as errors
make: *** [build/system/flashmem.o] Error 1
```

hreintke, 2015-12-11:
did you copy the files from the sdk_updates directory to sdk ?

alonewolfx2, 2015-12-11:
i copied now. here is new log. 

```cpp
D:\yedek\mb\Sming_RTOS\ESP_RTOS_POC>make
CC system/flashmem.c
In file included from include/user_config.h:4:0,
                 from system/flashmem.h:15,
                 from system/flashmem.c:1:
include/esp_systemapi.h:39:31: error: unknown type name 'ETSTimer'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
                               ^
include/esp_systemapi.h:40:30: error: unknown type name 'ETSTimer'
 extern void ets_timer_disarm(ETSTimer *a);
                              ^
include/esp_systemapi.h:41:29: error: unknown type name 'ETSTimer'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void *parg);
                             ^
include/esp_systemapi.h:41:42: error: unknown type name 'ETSTimerFunc'
 extern void ets_timer_setfn(ETSTimer *t, ETSTimerFunc *pfunction, void *parg);
                                          ^
system/flashmem.c: In function 'flashmem_erase_sector':
system/flashmem.c:100:3: error: implicit declaration of function 'WRITE_PERI_REG' [-Werror=implicit-function-declaration]
   WRITE_PERI_REG(0x60000914, 0x73);
   ^
cc1.exe: all warnings being treated as errors
make: *** [build/system/flashmem.o] Error 1

D:\yedek\mb\Sming_RTOS\ESP_RTOS_POC>make
CC system/flashmem.c
C+ appinit/user_main_example.cpp
In file included from c:/Espressif/ESP8266_RTOS_SDK/include/lwip/lwip/ip_addr.h:38:0,
                 from c:/Espressif/ESP8266_RTOS_SDK\include\espressif/esp_misc.h:28,
                 from c:/Espressif/ESP8266_RTOS_SDK\include\espressif/esp_common.h:95,
                 from appinit/user_main_example.cpp:12:
c:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:23: error: invalid suffix on literal; C++11 requires a space between literal and string macro [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                       ^
c:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:31: error: invalid suffix on literal; C++11 requires a space between literal and string macro [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                               ^
c:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:40: error: invalid suffix on literal; C++11 requires a space between literal and string macro [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                        ^
c:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:49: error: invalid suffix on literal; C++11 requires a space between literal and string macro [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                                 ^
cc1plus.exe: all warnings being treated as errors
make: *** [build/appinit/user_main_example.o] Error 1
```

hreintke, 2015-12-11:
Added two lwip updates, moves files to corresponding directories in sdk_updates

alonewolfx2, 2015-12-11:
ld files into the C:\Espressif\ESP8266_RTOS_SDK\ld and ets_sys.h into the C:\Espressif\ESP8266_RTOS_SDK\include\espressif right ?

hreintke, 2015-12-11:
no ets_sys in C:\Espressif\ESP8266_RTOS_SDK\include\espressif\esp8266
The filestructure in sdk_updates now reflects the directory of sdk

hreintke, 2015-12-12:
More investigaion : 
I found that by compiling all modules, also the "source directories" are handled as "include directories".
Just keeping this include definition, not compiling already breaks the application.

So now looking at include file interference

hreintke, 2015-12-13:
Solved.
Combination of makefile, several include file dependencies and IPv6 support implemented in RTOS

closed 2015-12-13 by hreintke