SmingHub/Sming #481: OSX Quick Start Guide Produce Error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope

anusoft, 2015-12-13:
I'm following https://github.com/SmingHub/Sming/wiki/MacOS-Quickstart and got this error:

```
In file included from SmingCore/Network/../../SmingCore/../SmingCore/Network/NtpClient.h:8:0,
                 from SmingCore/Network/../../SmingCore/SystemClock.h:7,
                 from SmingCore/Network/../../SmingCore/SmingCore.h:18,
                 from system/m_printf.cpp:10:
SmingCore/Network/../../SmingCore/../SmingCore/Network/../Platform/Station.h:36:25: error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope
  SCT_EspTouch_AirKiss = SC_TYPE_ESPTOUCH_AIRKISS
                         ^
make: *** [out/build/system/m_printf.o] Error 1
```

The prebuilt package https://bintray.com/artifact/download/kireevco/generic/esp-open-sdk-1.3.0-macos-x86_64.tar.gz missing some declares. You'll have to replace smartconfig.h with this typedef and add this line:

typedef enum {
    SC_TYPE_ESPTOUCH = 0,
    SC_TYPE_AIRKISS,
+++	SC_TYPE_ESPTOUCH_AIRKISS,
} sc_type;

+++bool smartconfig_set_type(sc_type type);

raburton, 2015-12-13:
Sming requires SDK v1.4.0/v1.5.0. It appears these instructions and binaries are out of date. @kireevco is the mac person in the team so perhaps he can update these...

kireevco, 2015-12-13:
@anusoft hey, can you try https://bintray.com/artifact/download/kireevco/generic/esp-alt-sdk-1.4.0-macos-x86_64.zip and https://bintray.com/artifact/download/kireevco/generic/esp-alt-sdk-1.5.0-macos-x86_64.zip ?

kireevco, 2015-12-13:
I've been working lately on a nice cross-platform package that helps all platforms

anusoft, 2015-12-13:
Both works but you'll need to symlink xtensa:
```
ln -s /opt/xtensa-lx106-elf/ /opt/esp-open-sdk/xtensa-lx106-elf
```
for the build would works with ESP_HOME=/opt/esp-open-sdk setting.

and to build Basic_Blink you'll need to chage ESP_HOME = /opt or you'll get an error if it set to default ESP_HOME = /opt/esp-open-sdk

```
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file ../ld/eagle.rom.addr.v6.ld: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1
```

kireevco, 2015-12-13:
@anusoft what if you set ESP_HOME=/opt/esp-alt-sdk ?

anusoft, 2015-12-13:
Same error only /opt works. I found some odds in xtensa-lx106-elf path which has duplicate contents in

xtensa-lx106-elf and xtensa-lx106-elf/xtensa-lx106-elf

Informatic, 2015-12-13:
I can confirm the exact same behaviour with invalid ld include path on Linux, both prebuilt esp-open-sdk 1.4.0 and esp-alt-sdk 1.5.0.

anusoft, 2015-12-14:
spiffy binary also doesn't build with this Make file

```
# Generating spiffs_bin
Checking for spiffs files
No files found in ./files.
/bin/sh: /opt/Sming/Sming/spiffy/spiffy: No such file or directory
Creating empty out/firmware/spiff_rom.bin
```

itsjustvenky, 2015-12-14:
Go into "spiffy" folder and use "make" command and then build app's or if you don't want SPIFFS, then set "export DISABLE_SPIFFS=1", make clean and then make

anusoft, 2015-12-14:
@itsjustvenky I knew but just to point out this issue since it should produce no error in example project.

Making Basic_Blink output spiffs path but none binary produce some of Spiff path but no the binary:
```
20:38:18 **** Build of configuration Sming for project SmingFramework ****
make all 
AS system/irq_check.s
CC system/flashmem.c
CC Services/FATFS/ff.c
CC Services/SpifFS/spiffs_cache.c
CC Services/SpifFS/spiffs_check.c
CC Services/SpifFS/spiffs_gc.c
CC Services/SpifFS/spiffs_hydrogen.c
CC Services/SpifFS/spiffs_nucleus.c
CC Services/SpifFS/spiffs_sming.c
CC Services/libemqtt/libemqtt.c
```

But manually make spiffy from /opt/Sming/Sming/spiffy does
```
CC spiffy.c
CC ../Services/SpifFS/spiffs_cache.c
CC ../Services/SpifFS/spiffs_nucleus.c
CC ../Services/SpifFS/spiffs_hydrogen.c
CC ../Services/SpifFS/spiffs_gc.c
CC ../Services/SpifFS/spiffs_check.c
LD spiffy

-rw-r--r--   1 anu  wheel    564 Dec 14 00:05 Makefile
-rw-r--r--   1 anu  wheel   3592 Dec 14 20:39 spiffs_cache.o
-rw-r--r--   1 anu  wheel  10624 Dec 14 20:39 spiffs_check.o
-rw-r--r--   1 anu  wheel   6748 Dec 14 20:39 spiffs_gc.o
-rw-r--r--   1 anu  wheel  12572 Dec 14 20:39 spiffs_hydrogen.o
-rw-r--r--   1 anu  wheel  23576 Dec 14 20:39 spiffs_nucleus.o
-rwxr-xr-x   1 anu  wheel  59996 Dec 14 20:39 spiffy
-rw-r--r--   1 anu  wheel   7137 Dec 14 00:05 spiffy.c
-rw-r--r--   1 anu  wheel   7592 Dec 14 20:39 spiffy.o
```

Informatic, 2015-12-14:
@anusoft I think that's expected, since `make all` is only building the framework, not "spiffy" tool. (as you can see in your listing)

Nevertheless, I think adding `DISABLE_SPIFFS = 1` in `Makefile-user.mk` of examples which do not require SPIFFS might be a good idea, since even blank spiffs image flashing takes a lot of time. I think that's a good candidate for a separate issue or PR :)

anusoft, 2015-12-18:
@Informatic Some examples will unable to use without Spiff, sure some projects might benefits to disable Spiff, but still need to fix spiff building in "make all" anyway.

raburton, 2015-12-18:
`make all` should not build spiffs (this has been discussed previously, multiple times), it is not required for all projects and building it adds a dependency on a host compiler.

anusoft, 2015-12-18:
@raburton I mean only example projects that need spiffy should build spiffy with "make all", not in every projects, sorry for misunderstanding.

My point is every examples should be able to build without problem.

matoushybl, 2016-01-05:
I've tried setting up Sming on my MacBook, but the compilation has failed with this error: `/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.1.0/../../../../xtensa-lx106-elf/bin/ld: cannot find -lsming` Do you have any clue what might be the problem? I've used SDK 1.5.

Thanks

hreintke, 2016-01-06:
@matoushybl : 
As mentioned in the install guides you need to build sming first

matoushybl, 2016-01-06:
I see, it wasn't mentioned in the OSX quickstart guide, so I've missed it. Shall I add it to the wiki with updated SDK as well?
Thanks.

hreintke, 2016-01-06:
@matoushybl : 
Yes please, and if you notice other mistakes/missing item can you add that too ?

Although I am not 100% sure that that doesn't require additional access rights.

raburton, 2016-01-12:
wiki appears to have been updated, thanks

closed 2016-01-12 by raburton