SmingHub/Sming #483: Compilation Error!

ashubhatt, 2015-12-14:
I am using the latest Unofficial SDK which provides ESPRESSIF's SDK v1.5

Console Output:
make all 
LD out/build/app.out
c:/Espressif/ESP8266_SDK/lib\libwpa.a(wpa_auth.o): In function `wpa_receive':
(.irom0.text+0x9a0): undefined reference to `aes_wrap'
c:/Espressif/ESP8266_SDK/lib\libwpa.a(wpa_auth.o): In function `__wpa_send_eapol':
(.irom0.text+0xd73): undefined reference to `aes_wrap'
collect2.exe: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

Any solution ?
I am new to Sming, forgive me if this problem is already addressed.

hreintke, 2015-12-14:
@ashubhatt : 
If you use the Sming 1.4.0 (= latest release release = master branch) then highest SDK release is 1.3.0.

We are close to releasing next version (already available in Develop branch) which supports SDK 1.4.0 and 1.5.0 (not 1.3.0)

ashubhatt, 2015-12-14:
Was using an older version of Sming!

ashubhatt, 2015-12-14:
I am getting this error:

make: esptool2: Command not found
make: *** [out/build/app.out] Error 127

I have already compiled and placed the esptool2 folder in the Sming directory. But the compilation still shows above error. What could be the problem?

hreintke, 2015-12-14:
You did set the environment variable esptool2 to point to the esptool2.exe
 
You can also set the value in your makefile like : 
ESPTOOL2 ?= c:\userdata\esp8266\esptool2\esptool2.exe

ashubhatt, 2015-12-14:
I did set the variable but it was incorrect; only added the path not the .exe file!

Thank you. Finally its working perfectly.

closed 2015-12-14 by ashubhatt