SmingHub/Sming #484: Basic_Boot fails to build because of changes to os_malloc

GJMontreal, 2015-12-15:
I recently, i.e. this morning, tried building the Basic_rBoot example using V1.3 of the esp sdk. It fails building /rboot-api with too few arguments to function 'pvPortMalloc' etc.
 I noticed on http://www.esp8266.com/viewtopic.php?f=34&t=5748 that there was a remark saying that the underlying implementations of the memory allocation routines have changed. 
Am I alone in thinking that rBoot support in Sming is currently broken?

raburton, 2015-12-15:
The changes you mention were made in sdk v1.4.0 (as also stated in the link you provided). You say you are building with sdk v1.3.0 so this doesn't make sense, as you would not get this error with sdk v1.3.0. To build with sdk v1.4.0/v1.5.0 you need to use the development branch of Sming (or the upcoming  Sming v2.0, once it is released).

GJMontreal, 2015-12-15:
Thanks for your help. You're absolutely right, it doesn't make sense and you just sent me off in the right direction. Indeed I am building with the V1.3 sdk, but I'm on the develop branch of Sming!! Oops. Switched to master and sure enough, it builds without problem. Thanks again.

closed 2015-12-15 by GJMontreal