SmingHub/Sming #485: Can't build any example on Windows

Bravo13, 2015-12-15:
Trying to run latest stable(just installed from chocolate). So - in Sming dir runned make - all is ok. Now cd to C:\tools\sming\Basic_Serial run make clean && make Got error c:/Espressif/ESP8266_SDK/lib\libwpa.a(wpa_auth.o): In function `wpa_receive': (.irom0.text+0x9a0): undefined reference to `aes_wrap' c:/Espressif/ESP8266_SDK/lib\libwpa.a(wpa_auth.o): In function `__wpa_send_eapol': (.irom0.text+0xd73): undefined reference to `aes_wrap' collect2.exe: error: ld returned 1 exit status make.EXE": *** [out/build/app.out] Error 1
What I'm doing wrong?

raburton, 2015-12-15:
Looks like you are using sdk v1.5.0, Sming stable only supports sdk v1.3.0. To use a newer sdk use the development branch of Sming.

Bravo13, 2015-12-15:
Seems you are right. Renaming ESP8266_SDK to ESP8266_SDK_150 and ESP8266_SDK_130 to ESP8266_SDK solves problem. May be it would be better to fix it in choco repo?

kireevco, 2015-12-15:
This is a part of UDK. Choco is just pulling things from UDK source.

hreintke, 2015-12-15:
UDK has several releases in the repo.
Can we point the Choco dowload to the UDK 2.0.8 version ?
Then it is compatible with the latest Sming version.

raburton, 2015-12-17:
> Can we point the Choco download to the UDK 2.0.8 version?

Aren't we about to replace UDK?

hreintke, 2015-12-17:
@raburton : @kireevco : 
I know we are busy getting a new SDK release out but..
- It is work in progress 
- Also for the new SDK we should have the principle that if a user follows the installation procedure from github/released version, he/she should get a "Release compatible" version and not the latest available.

kireevco, 2015-12-18:
I updated `esp8266-udk` packages, `choco install esp8266-udk` installs 2.0.8 and not 2.0.9 (which has espressif sdk 1.5.0 by default. Please give it a try.

closed 2015-12-20 by hreintke