SmingHub/Sming #488: Multiple callbacks and Serial receive causes reset/crash

piperpilot, 2015-12-16:
I've been fighting this one for a couple of days now.  I have narrowed it down to having multiple callbacks and using serial to receive with either a callback or a delegate.  I'm not sure if there are other instances where it happens as well.  The attached, modified version of HttpServer_ConfigNetwork shows the problem.  In order to reproduce, do the following:

1.  Compile with latest develop and SDK 1.4 or 1.5
2.  Flash to your ESP and run it
3.  Connect to the serial terminal at 19200.  Once the timer callback has finish as noted by receiving some characters like:

SBIP192.168.1.14;
SBAP0;
SBID16666831;

Send some strings to the serial.  It is looking for a null terminated string.  In my testing, I send the following as it is JSON and thats what my project uses:

{"time":"00:00:30","grl":210,"fd1":999,"fd2":999,"err":-422,"p":-99,"i":0,"d":0,"dpr":0,"hs":0,"ld":0,"set":250,"ds":1,"sot":0,"kp":100,"ki":30,"kd":100}

It normally will crash within 4-10 sends of the data string.  You don't need to send the data fast, once every second or two will do it.

It seems much more likely to happen when you connect it to your wireless...to do that, search for the access point "Smobot Wifi" and connect.  Then point your browser to 192.168.4.1.  From there, connect to your home wifi and reset. 

If I roll back to commit 77d3dfdfb68ea35a542ed3628cba75eb175afda1 and SDK 1.3 everything works fine and it doesn't crash.

[HttpServer_ConfigNetwork.zip](https://github.com/SmingHub/Sming/files/64649/HttpServer_ConfigNetwork.zip)

hreintke, 2015-12-17:
@piperpilot : 
Can you test with #489 merged into your project ?

In there I changed the handling of Serial Delegate & CommandProcessing within the HardwareSerial Class. 

PS : In this patch the availableCharsCount parameter of the delegate does not yet contain the actual count but always is one (1).

PS2 : I just verified correct behavior in my local environment and not with your HttpServer_ConfigNetwork.zip

piperpilot, 2015-12-17:
I did a quick check on this and it looks to fix the problem. I will test further in a little while this really needs to make the 2.0 release.

hreintke, 2015-12-19:
I updated the PR to a complete version, supporting all current functionalities of serial delegate & commandprocessing.

@piperpilot : Can you verify correctness in your test environment ?

piperpilot, 2015-12-21:
@hreintke This seems to be a good fix.  I was able to crash the serial once, but I think that was due to buffer overflow which is fixed in another PR.

hreintke, 2015-12-21:
Thanks for testing.
I will merge this PR into develop.
The other is also merged this morning, so should be all correct  now

closed 2015-12-21 by hreintke