SmingHub/Sming #491: LWT feature of MQTT not working

ashubhatt, 2015-12-17:
I am using Sming 1.4.0 and trying the sample code "MQTTClient_Hello".
Although the LWT function is implemented, the actual LWT message is not received in case of abrupt disconnection.

I have opened issue #337, but it was closed assuming the latest build have added the feature; the bug still remains.

Kindly update the source code.

Informatic, 2015-12-27:
I can confirm it is working now.

matoushybl, 2016-02-16:
Is LWT working with the patch? After setting LWT and disconnecting power, it was still not sent.

flexiti, 2016-02-16:
Are You using last develop ?

matoushybl, 2016-02-17:
Yes, I do. Is the keep alive interval important for LWT? I mean if it is for example 1 hour, will it send LWT after the one hour?

flexiti, 2016-02-18:
no, keep alive is not important for LWT if socket is closed eg. no internet connection
(depends on broker)
For me LWT is working after aprox. 30 sec then device is off

> if(!mqtt.setWill("main/device/100","{\"Value\":0,\"No\":100,\"Status\":\"Off Line\",\"Time\":\""+SystemClock.getSystemTimeString()+"\"}", 1, true)) {
> 			debugf("Unable to set the last will and testament. Most probably there is not enough memory on the device.");
> 		}
> 
> 	mqtt.connect("esp8266-100",MQTT_USERNAME,MQTT_PWD);
> 	mqtt.subscribe("main/#");
> 	mqtt.publishWithQoS("main/device/100","{\"Value\":1,\"No\":100,\"Status\":\"On Line\",\"Time\":\""+SystemClock.getSystemTimeString()+"\"}",1,true);
>

matoushybl, 2016-02-18:
I have a very similar code as yours, I am starting to think that this is caused by broker, at my second attempt the LWT was received after more than two hours. What broker are you using?

flexiti, 2016-02-18:
Mosquitto on my serwer

matoushybl, 2016-02-18:
Thanks, me too. That is weird, I updated mosquitto and it still isn't working, I'll dig deeper to find out what the problem is.

matoushybl, 2016-02-23:
Yeah, by power off, I haven't noticed the question, sorry.

hreintke, 2016-03-25:
Fixed

closed 2016-03-25 by hreintke