SmingHub/Sming #492: How to check if WiFi gets disconnected?

ashubhatt, 2015-12-18:
Hello Developers,

Suppose ESP is operating as Station and if WiFi gets disconnected during normal operation, I would like to implement some action on that event.
I saw the APIs of Station.h but could not find one.

Kindly point me to the right direction.

flexiti, 2015-12-18:
StationClass::isConnected()

alon24, 2015-12-18:
Does the station fire an event when disconnecting or connecting for the
second time?

ashubhatt, 2015-12-18:
i dont think any event is fired. The StationClass::isConnected() needs to be called manually.

May be a continuous timer function needs to be implemented that checks the connection periodically.

hreintke, 2015-12-18:
Wouldn't IP connection be closed & callback fired when wifi disconnects ?

flexiti, 2015-12-18:
It seems to me that the easiest way every few seconds (eg. a timer) to check the network status. Additionally, you can then check a lot of things.
Normally esp runs automatically reconnect, but I understand you want to know when network was off ,or poor

ashubhatt, 2015-12-18:
@hreintke  I cannot find any callback function which fires upon closed IP connection. If there is one, kindly provide the function to set the callback function.

@flexiti Yes for the temporary solution, I have implemented a timer which checks StationClass::isConnected(). If found disconnected, appropriate actions will be taken.

hreintke, 2015-12-18:
@ashubhatt : I was half wrong/half right.
On a client connection you have the HttpClient/TcpClient...CompleteDelegate but not on a server, due to the nature of "waiting for an incoming connection.

I checked documentation -> There is a system callback on connect, wifi_set_event_handler_cb
Page 75 of SDK programming guide.
If you make an adaption to Station class setting and using delegate you can submit a PR for inclusion in Sming.

ashubhatt, 2016-01-28:
@hreintke Thanks for the information. It saved my one precious timer.
Kindly merge #562, so I can send new PR for this feature.

ashubhatt, 2016-01-31:
Hello @all, in my opinion we can clean the Station class and update it to remove the following functions:

`void waitConnection(ConnectionDelegate successfulConnected);` and
`void waitConnection(ConnectionDelegate successfulConnected, int secondsTimeOut, ConnectionDelegate connectionNotEstablished);` and
`void internalCheckConnection();` and
`static void staticCheckConnection();` and
`ConnectionDelegate onConnectOk;` and
`ConnectionDelegate onConnectFail;` and
`int connectionTimeOut;` and
`uint32 connectionStarted;` and
`Timer* connectionTimer;`

These functions are implemented to simply set callbacks for WiFi Connection and Disconnection using a timer to redundantly check for connection parameters every 50ms. However as @hreintke pointed out, we can instead use `wifi_set_event_handler_cb` function to let the SDK handle the connection events.

Moreover these functions can be simply replaced by:
`ConnectionDelegate onDisconnect;`
`ConnectionDelegate onConnect;`

`void internalWifiEventHandlerCallback(System_Event_t *event);`
`static void staticWifiEventHandlerCallback(System_Event_t *event);`
`void setWiFiConnectCallback(ConnectionDelegate connectCb);`
`void setWiFiDisconnectCallback(ConnectionDelegate disconnectCb);`

hreintke, 2016-01-31:
@ashubhatt : 
Agree that Station class can be improved using the events provided by the SDK.
However I would like to keep compatibility with the current API, even if that gives some similar functions.

If you provide a PR with a base implementation of these improvements we can discuss from that.

BTW : The same can be done with AccessPoint Class.

ashubhatt, 2016-01-31:
@hreintke I have implemented and tested this functionality from Station Class, but not with AccessPoint Class but now since you have pointed out, I will make the necessary changes in that class too.

I can definitely provide a PR, but I dont know how I can until my previous PR is merged.

hreintke, 2016-01-31:
@ashubhatt : 
Although I fail to see why the other PR is related to this one, it is merged, :smile:

anakod, 2016-02-01:
Timers in **waitConnection** was required in Sming from begining while this events API isn't exist on ESP (added after SDK 1.1, I think). Now Espressif implemented it (and have fixed many bugs inside :smile: ) so we can remove old code, but with full back compatibility for end users. It's why I like Sming approach where users didn't dependent on internal API implementation and availability.

Also will be fine to check is events availabe on RTOS\ESP32 to keep future compatibility.

avr39-ripe, 2016-03-03:
#611 AFAIK can resolve this for both station and accesspoint.

hreintke, 2016-03-03:
@all : 
Agree, solved by #611

closed 2016-03-03 by hreintke