SmingHub/Sming #496: How to get proper callback

Bravo13, 2015-12-21:
Is it possible to do something like this. So that I can use responce in WifiStation.startScan callback
```c
void httpAsyncList( HttpRequest &request, HttpResponse &responce){
    WifiStation.startScan([&responce](bool success, BssList list)mutable{
        JsonObjectStream* stream = new JsonObjectStream();
        JsonObject& json = stream->getRoot();
        json["scanSuccess"] = (bool)success;
        responсe.sendJsonObject(stream);
    });
}
```

hreintke, 2015-12-22:
@Bravo13 : 
It is better to formulate in more generic terms what you try to achieve. By just posting a difficult to understand piece of code, not many people will be able to help.

Bravo13, 2015-12-22:
Well, in generic, I would like to run startScan and get result of it in one request. So I need to send response asynchronously. To archive it, I need to use response object in startScan callback.

hreintke, 2016-01-27:
@Bravo13 : Little late but yet an answer.
There is no possibility in current sming which provides this possibility. 
The startscan is an function from the sdk and only available with callback.
And there is no possibility for a "delayed httpresponse" 

There might be options to add functionality like this but it is not on the TODO list now.

closed 2016-03-25 by hreintke

rickyitexpert, 2016-06-19:
I guess this can be done asynchronously using websocket and java scripting