SmingHub/Sming #497: delay() in procTimer causes reset

anusoft, 2015-12-21:
Reset Message:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 23116, room 16 
tail 12
chksum 0xa1
ho 0 tail 12 room 4
load 0x3ffe8000, len 884, room 12 
tail 8
chksum 0x37
load 0x3ffe8378, len 1944, room 0 
tail 8
chksum 0x5d
csum 0x5d
don't use rtc mem data
rlS
```

Code that trigger
```
Timer procTimer;

void blink()
{
	delay(1000);
}

void init()
{
procTimer.initializeMs(1000, blink).start();
}
```

Testing on nodemcu V1

flexiti, 2015-12-21:
This is not Arduino, Sming is event driven .. don't use delays > 50 ms

hreintke, 2015-12-21:
You can look at the Basic_Blink example to implement "espressif/sming" like.

alonewolfx2, 2015-12-21:
@anusoft we dont like delay things in sming but your main problem is timing. 
you are starting timer with 1000ms interval and you are using 1000ms delay in timer function. basically system trying to start your blink function every second but your blink function stopping system. you can try 1000ms interval with 500ms delay and it will work fine.
ps:espressif says you cant use delay with more than 5second because of wdt.

anusoft, 2015-12-21:
@alonewolfx2 OMG, you're write. my mistake closing this thread now.

Thank you all.

closed 2015-12-21 by anusoft

hreintke, 2015-12-21:
@anusoft : 
It is not only the timer interval.

ANY delay() in ANY callback will cause issues and unexpected behavior.
And Sming processing is completely callback based.

flexiti, 2015-12-21:
@alonewolfx2 
>ps:espressif says you cant use delay with more than 5second because of wdt.

There is no need to use Delay() in the program, unnecessary consumption of CPU time .. by doing nothing.
From Espressif:

>‣ If a task occupies the CPU too long, ESP8266 can't feed the dog, it will cause a watchdog
reset (> 1s)；
‣ If interrupt is disabled, CPU can only be occupied in us range and the time should not be
**more than 10 us**; if interrupt is not disabled, it is suggested that CPU should not be
occupied more than **500 ms**.
• We suggest using a timer to check periodically, if users need to call os_delay_us or function
while, or function for in timer callback, please do not occupy CPU more than 10 ms.
• We suggest using RTOS SDK, RTOS to schedule different tasks.
Please do not call "os_delay_us","Delay", or "while" or "for" **to occupy CPU more than 10 ms in timer callback**.

Failure to comply with these rules makes at least TCP will not work, and this is after all the biggest advantage of ESP8266