SmingHub/Sming #501: I2C support for OLED1306 display without reset pin

harry-boe, 2015-12-22:
I played wit the new timer based version of the OLED 1306 example project.
After heaving many issues with the older delay Version I cn confirm that it's now working fin for SPI and I2C displays.

There is just one add on. Most of the I2C OLED display do not have a RESET pin.
Proper initialisation of those devices require another constructor:

-> current example:  

    // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
    display.begin(SSD1306_SWITCHCAPVCC);`

This works for I2C with RESET and SPI 

-> I2C without reset needs to be initialised like that:

    // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)`
    // initialize with the I2C addr 0x3D (for the 128x64)
    // bool:reset set to TRUE or FALSE depending on you display
    display.begin(SSD1306_SWITCHCAPVCC, SSD1306_I2C_ADDRESS, FALSE);`

Maybe we add some comments in the code to explain i2c support

alonewolfx2, 2015-12-22:
Good catch. Can you make pr ?

sle118, 2015-12-29:
Wow! This will save me some time! I actually captured the trace of SPI communication from the initialization to understand why the display wasn't firing up, and I was about to start analyzing. so thank you!

harry-boe, 2016-01-05:
as requested pr made: now set to I2C as default but SPI still there in comments.

hreintke, 2016-01-06:
@harry-boe : 
I did not read all the details but you mention : "now set to I2C as default"
Suppose that will break code which depends on the old default,

Is so, you should revert to the current default.

harry-boe, 2016-01-06:
Should not break anything. There are no changes in the libs. 
The change is only in the sample application.cpp

maybe default was the wrong wording here ..

harry-boe, 2016-01-27:
THX for merging

closed 2016-01-27 by harry-boe

fvalenta, 2018-07-19:
Works through I2C without reset. The undocumented 0xe4 command has to be sent.

destroyedlolo, 2018-12-25:
I'm working on the port of this library to unix SBC (like RaspberryPI, BananaPI, and all other *PIs) : see https://github.com/destroyedlolo/ArduiPi_OLED
I'm experiencing some issues with my Display turn off with 0xAE command but never tune on back with 0xAF one.
Do you think it can be solved with this 0xE4 command ? Where have I to add it ?
Thanks

feer9, 2019-04-08:
> Works through I2C without reset. The undocumented 0xe4 command has to be sent.

Hi, can you please tell, how to use that 0xe4 command? where do I put it? at the beginning of initialization?

vjalle, 2022-01-09:
Hi,
The 0xE4 does not reset the module. I did not connect the reset line, and sometimes after powerup (or after intentionally disturbing the I2C communication by sticking a screwdriver between the pins) the display was not working (went blank, previous content froze, or got a random bit pattern). It acknowledged I2C commands, but the display was frozen. As an experiment, with the hope that I'll find an undocumented reset command, I started sending it all undocumented commands with different data (and with not data, too), and observed that after sending 0xE4, the display started working properly. Sending 0xE5 (or 0xE3 - can't remember) had the opposite effect, that froze the display.
Added 0xE4 to the init sequence, and the problem seems to have been solved. Did not have frozen display on powerup since adding this fix, but I can still provoke empty display with the cursed screwdriver. That might be my code failing to detect the communication error and restarting the init sequence (?).
The chip seems to be in random state after powerup, and broken communication can also upset the internal registers. I assume that writing all registers with the proper content, including issuing the undocumented commands will restore normal operation, so a reset might not be needed.