SmingHub/Sming #504: right place to start WifiStation.waitConnection

avr39-ripe, 2015-12-23:
Hello! After some SDK change WifiStation.waitConnection do not work for me as expected - if I invoce it in such order
```c++
	WifiStation.enable(true);
	WifiStation.config(ActiveConfig.StaSSID, ActiveConfig.StaPassword);
        WifiStation.waitConnection(StaConnectOk, StaConnectTimeout, StaConnectFail);
```
and if esp can't connect to ActiveConfig.StaSSID wifi network method StaConnectFail never called. Instead of this there are meny "...recconect after 1 s" messages than came forever. If network is reachable, and esp8266 successfuly conects to it all works as expected and StaConnectOk called.
After some experiments and investigation I found that if I just change the order ov functions call all works as expected in BOTH cases: either accessible or inaccessible wifi network. So here is correct order:
```c++
	WifiStation.waitConnection(StaConnectOk, StaConnectTimeout, StaConnectFail);
	WifiStation.enable(true);
 	WifiStation.config(ActiveConfig.StaSSID, ActiveConfig.StaPassword);
```
Do please comment on this, test, confirm or suggest better/cleander solution.

tprochazka, 2016-01-06:
Just small note. You don't need to call
WifiStation.config(ActiveConfig.StaSSID, ActiveConfig.StaPassword);
in every time in init(), Espressif SDK remember last value itself.

avr39-ripe, 2016-03-03:
This will be resolved by #611 which in fact changes the way of handling wifi related events.

closed 2016-03-03 by avr39-ripe