SmingHub/Sming #508: Sming support on http://platformio.org/

tprochazka, 2015-12-27:
Will be really great for Sming if it will have support on http://platformio.org/#!/
Espressif SDK is already there and Arduino implementation too
http://platformio.org/#!/boards?filter%5Bframeworks%5D=arduino

hreintke, 2015-12-31:
@tprochazka : 
I understand the enhancement but I have no idea what needs to be done and how difficult it would to maintain.
Would suggest that you investigate further on the above and/or other aspects and feedback.
Based on that we can decide on further actions and if neeed I will expect a PR from you.

tprochazka, 2015-12-31:
I just found that it exists, I will try to use it and investigate how difficult it is.

tprochazka, 2016-01-01:
Here in video presentation how it works
https://www.youtube.com/watch?v=lXchL3hpDO4

nistvan86, 2016-01-09:
+1
Sming is an amazing project but as a beginner with Eclipse CDT projects I've found it hard to create a new project from scratch. PlatformIO greatly simplifies this process.

chriscohnen, 2016-03-08:
+1
I am also interested in having Sming available on PlatformIO
Hope this will be implemented

hreintke, 2016-05-23:
I agree on the suggestion to allow sming installation with platformio.
Can someone take the lead in that.
Please open an issue on SmingRTOS on that.

closed 2016-05-23 by hreintke