SmingHub/Sming #509: Cannot delete JsonObjectStream to free memory

kingoflolz, 2015-12-27:
I have a function which checks the state of some GPIO pins, puts them in a JSON object and then sends it to the client. It works fine without the last line except that it eats 200 bytes of memory every time it is run. 
```c++
void led_api_state(HttpRequest &request, HttpResponse &response) {
    JsonObjectStream* stream = new JsonObjectStream();

    JsonObject& json = stream->getRoot();

    JsonArray& leds = json.createNestedArray("leds");

    JsonObject& led0 = leds.createNestedObject();

    led0["led"] = 0;
    led0["state"] = digitalRead(5);

    JsonObject& led1 = leds.createNestedObject();

    led1["led"] = 1;
    led1["state"] = digitalRead(4);

    response.setAllowCrossDomainOrigin("*");
    response.sendJsonObject(stream);
    //works fine without next line, but eats memory
    delete stream;
}
```

None of the examples use the delete statement but in #190 it was recommended.

hreintke, 2015-12-27:
@kingoflolz : 
What version of Sming are you using ?
In Sming 1.4.0 there is a known issue with memoryleaks in json, solved in V2.0

kingoflolz, 2015-12-27:
@hreintke I'm running the code on the development branch, should be 2.0.0

hreintke, 2015-12-27:
@kingoflolz : 
Reading the initial issue again, I don't understand your issue.
In which situation do you have the memory leak ?

kingoflolz, 2015-12-27:
@hreintke If I do not delete the JsonObjectStream, it uses 200 bytes of memory every time it is run and does not free it, but otherwise works. If i try to delete the JsonObjectStream, it will make the ESP hang and cause a watchdog reset.

hreintke, 2015-12-28:
@kingoflolz : 
Your code is NOT identical to the one described in #190.
You have a response.sendJsonObject(stream); which takes care of deleting stream.

On memory usage. What you are seeing in the usage of the underlying TCP network. Due to the nature of the TCP, some buffers are only deleted after some time.
This is the result on my system : 
```
TCP connection closing
~TCP connection
New Heap size: 38648, diff = 7500
-TCP connection
New Heap size: 40280, diff = 500
New Heap size: 40112, diff = 10000
New Heap size: 40072, diff = 24500
New Heap size: 40112, diff = 500
New Heap size: 40296, diff = 76500
New Heap size: 40480, diff = 10000
New Heap size: 40664, diff = 2500
New Heap size: 40848, diff = 3500
New Heap size: 41032, diff = 4500
New Heap size: 41248, diff = 7500
```
The diff is the number of milliseconds between prints.
The "nice values of diff" are because I check very 500 ms for a heap change.

kingoflolz, 2015-12-29:
@hreintke Thank you for figuring out the issue, but is there any way to make the TCP stack free its memory faster? Because I am getting memory exceptions on the ESP from having AJAX requests running in a loop.

hreintke, 2015-12-29:
@kingoflolz : No as far as I know the is no option to return memory faster.
I think it returns memory after two minutes. Limitation is then the amount of requests coming in in that period.

Closing this issue as the original question is solved.

closed 2015-12-29 by hreintke