SmingHub/Sming #510: Websocket Client did not merged into Releast 7.0

williamesp2015, 2015-12-27:
@hrsavla spent efforts to make Websocket Client for Sming and added pull request but did not imported into Sming. It is very good if someone add binary send and receive to the current project.
https://github.com/hrsavla/Sming/tree/15549097b0681b79b1edc70e4cb27cc9b913333e/Websocket_Client

robotiko, 2015-12-27:
@williamesp2015 is it working?
Tested?
I recall there were changes to be done to adapt to sming style but never done.

williamesp2015, 2015-12-31:
Does anyone like @hrsavla or @ADiea can work on this to be successfully work on Sming?

closed 2015-12-31 by williamesp2015

ADiea, 2015-12-31:
@williamesp2015 not on my priority list right now sorry.
I am preparing a major commit to bring lwip compiled from sources to sming, to be able to see traces from the tcp/ip stack etc bc we have a lot of misterious bugs related to network

hreintke, 2015-12-31:
@ADiea : 
We are moving Swing from NOSOS SDK to RTOS SDK.
In my opinion we should not put effort in the LWIP of Sming NONOS.

Within the conversion process I had some issues with LWIP and tried to use the NONOS LWIP from Espressif in RTOS SDK but failed. Also noticed then that NONOS and RTOS LWIP implementations differ. There is no (yet?) LWIP for RTOS from espressif.
There is a LWIP in the esp-open-rtos project. I was not able to use that either.

I stopped the LWIP compile effort when I solved the Sming RTOS LWIP issue.
That was related to the LWIP IPv6 inclusion of  RTOS LWIP

ADiea, 2015-12-31:
@hreintke 
Why not have 2 versions of SMING one NONRTOS (mature let's say) and the new RTOS version

I have succeded in compiling everything up for the NONOS version os sming, will submit a PR
Also my tests show somehow a more stable implementation than just linking with liblwip.a as it's currently done in sming NONOS

Also, how is the new RTOS? Did you do any tests yet? 
1. how much heap you are left with, 
2. how many tasks/processes you can have
3. lag compared to nonrtos
4. is it stable? based on freertos?
etc

Cheers

hreintke, 2015-12-31:
@ADiea : 
Supporting two versions of Sming is not useful, it takes additional effort of scarce resources

For moving to RTOS SDK : 
- Espressif is moving support from NOSOS to RTOS -> will keep Sming future proof
- Opens the Sming Support for ESP32
- Based on FreeRTOS which will provide tasking/threads -> Standards based
- Opens better Sming support for "Arduino like" programming paradigms (without getting WDT resets).

And yes/no about testing. See #495 For information.
There is a "Working Proof of Concept" version which include most of the current functionality.
With this version the question you state will be answered.

If you have time, would appreciate your help in testing and maturing the RTOS version.