SmingHub/Sming #513: How to disable debug statements printing on Serial port?

ashubhatt, 2015-12-29:
Hey guys,

I have been trying to use the Serial port for connection with a relay controller. However, there are many debug statements which gets printed during WiFi and TCP connection.

I have already tried:
Serial.systemDebugOutput(false); 

But it doesn't seem to work. I don't want Serial monitor to print any debug statements.
Any solution??

fcgdam, 2015-12-29:
Did you try also by adding:

system_set_os_print(0);

?

flexiti, 2015-12-30:
if You have at the startup:
>eg. 
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

 it is the ROM log (startup log) You can't remove or disable it . But you could call system_uart_swap to use MTCK & MTDO as UART0 

See also:
http://bbs.espressif.com/viewtopic.php?f=6&t=610&p=2317&hilit=debug+off#p2317

ashubhatt, 2015-12-30:
Thank you guys for your inputs. So I tried system_set_os_print(0); and as @flexiti pointed, the ROM log is still printing. Seems its not possible to completely shut down debug logs.

alonewolfx2, 2015-12-30:
yes. there is no way to disable sdk level uart output yet. this is related directly espressif sdk.i am closing issue.

closed 2015-12-30 by alonewolfx2

jimsparkman, 2016-01-08:
This appears to work:
```C
Serial.begin(SERIAL_BAUD_RATE);
Serial.systemDebugOutput(false);
delay(10);
system_set_os_print(0);
delay(10);
system_uart_swap();
delay(10);
```