SmingHub/Sming #516: Add a TCP Disconnect Callback function

ashubhatt, 2016-01-01:
Hey guys,

Please provide a delegate to add a callback function for TCP disconnection. The callback function should fire once the functionTcpConnection::close() is fired.

If you guys can provide a workaround solution, I am all ears.

Thanks

ashubhatt, 2016-01-01:
Used: 
virtual err_t onReceive(pbuf *buf);

closed 2016-01-01 by ashubhatt

alonewolfx2, 2016-01-01:
@ashubhatt what is your full function for connection close event?

ashubhatt, 2016-01-01:
I used a delegate function for callback:
```
typedef Delegate<void()> MqttDisconnectCallback;
```

Then set this callback using a new function:
```
void MqttClient::setDisconnectCb(MqttDisconnectCallback dcb /* = NULL */){
	if(dcb)
		this->disCallback = dcb;
}
```

Finally in virtual err_t onReceive(pbuf *buf), I called the delegate function:
```
if (buf == NULL)
	{
		// Disconnected, close it
		this->disCallback();
		TcpClient::onReceive(buf);
	}
```

alonewolfx2, 2016-01-01:
ok. can you make PR for this?

ashubhatt, 2016-01-01:
The implementation was applied specifically for MQTT. But can be used similarly for other child classes of TCP Connection.
PR: #518