SmingHub/Sming #521: Problem with TemplateFileStream

dseliskar, 2016-01-01:
Happy New Year to everybody!

First thing I must say that I just started tinkering with Sming and it's great!
But I stumbled on a small issue with templates. In case variable is included
in template in such a place that TemplateFileStream::readMemoryBlock would start 
with it, then TemplateFileStream::readMemoryBlock will change state to eTES_Found 
and return 0. It relies on ::seek to change state to eTES_StartVar, but since it 
returns 0 the ::seek never gets called so it is stuck in a eTES_Found state returning 
0 over and over again. This occurs when variable is at beginning of file or is "split"
by FileStream::readMemoryBlock in which case TemplateFileStream::readMemoryBlock
starts with variable at the beginning and same situation appear.
As proof-of-concept fix I added oneliner (p1) (I know very inefficient - 
we calculate cur - tpl for the third time). It changes state to eTES_StartVar
immediatelly. I'm not sure if the fix is in the right place, but it somehow resolves 
the situation. Can somebody more into it look it over please.
Next thing I noticed is misspelled stream type eSST_TepmplateFile which should be fixed for
beauty of the th^H^Hsming. Thanks and keep the good work!

[p1.txt](https://github.com/SmingHub/Sming/files/75977/p1.txt)
[p2.txt](https://github.com/SmingHub/Sming/files/75978/p2.txt)

dseliskar, 2016-01-01:
LOL jobidon just beat me to it ! Please join the two.

hreintke, 2016-01-01:
@dseliskar : 
Thanks for "finding the issue" and then also providing the solution.
Can you submit a PR for this for further discussion an integration ?

hreintke, 2016-01-10:
solved by #522

closed 2016-01-10 by hreintke