SmingHub/Sming #524: TelnetServer Bug: curClient not initialized in constructor

gschmottlach, 2016-01-02:
I ran across a bug in TelnetServer that appears sporadically. The ```curClient``` member variable is **not** being initialized to NULL in the constructor. As a result, depending on what's stuffed in the memory region at start-up, you may not be able to telnet into your application. This fix is pretty simple. Here's the revised constructor with ```curClient``` explicitly initialized:

```
TelnetServer::TelnetServer()
	:
	TcpServer(),
	curClient(nullptr)
{
	// TODO Auto-generated constructor stub

}
```

hreintke, 2016-01-06:
@gschmottlach :
Good catch, thanks.
Did a quick check and think it is also valid for commandExecutor.

I prefer to have the intialization done in the TelnetServer.h where the variable is defined.

Can you make a PR for this ?

gschmottlach, 2016-01-07:
As requested I've submitted a PR for this. Both ```curClient``` and ```commandExecutor``` are set to ```nullptr``` in TelnetServer.h.

hreintke, 2016-01-31:
Solved

closed 2016-01-31 by hreintke