SmingHub/Sming #526: Additional cpp class inside app

andriyadi, 2016-01-03:
Call me ignorant, but I can not find any reference and make it work.
I add a new class (h and cpp) inside `app` folder. When I use the class and instance methid and build, it says: 
```undefined reference to MyClass::myMethod()```

Tried to put the class inside `include` folder, same result.

Please advice.

andriyadi, 2016-01-03:
Nevermind. I think the problem was that I use project template from sample of Sming RTOS POC, instead of from original Sming repo. Copying from existing Sming sample projects, I can add additional C++ class (.h & .cpp).

Moral story here, I'm dying to get Sming to support FreeRTOS :)
Apparently [Sming RTOS POC](https://github.com/hreintke/Sming_RTOS_POC) is still very early, which is understandable. One issue is like I said above, cannot add new C++ class, beside the default `application.cpp`.

hreintke, 2016-01-04:
@andriyadi : 
When using the RTOS Sming I can add my own class (cpp + h) in the app directory and use it in application. In the cpp you use MyClass::myMethod() and not myMethod() ?

RTOS Sming does already contain most of the NONOS Sming functionality. Which are you missing to start working with that ?

raburton, 2016-01-05:
Works fine on both nonos and rtos version and user now has it working for him too.

closed 2016-01-05 by raburton

andriyadi, 2016-01-06:
Apparently it's all because of my stupidity in creating the class. So it's all good now.

As side note, the initial cause I have the so-called issue is due to SMING RTOS doesn't have SPI hardware class, only SPISoft, so I need to do something about it that eventually lead to my raising issue. But I just noticed SMING RTOS is now having SPI class, so it's all good.

Keep it up the awesome work!

hreintke, 2016-01-06:
@andriyadi : 
As you noticed the SPI class is included recently.
Please provide feedback on the results of you using the class.