SmingHub/Sming #528: MQTT usename and password too short for IBM iot service

ericmnel, 2016-01-04:
IBM's iot service uses MQTT with a password length of 14 and a random token password (no idea how long the longest one is, but mine was 18 characters.  

In file /opt/Sming-2.0.0/Sming/Services/libemqtt/libemqtt.h

in the top of the file I propose changing these two constants:

    #define MQTT_CONF_USERNAME_LENGTH 13
    #define MQTT_CONF_PASSWORD_LENGTH 13

to be 20 for the username and 30 for the password.  15 and 19 would be the minimum but longer would be safer.

etmmahi, 2016-01-04:
in libemqtt.h these values are only defined when not defined on a global level
```
#ifndef MQTT_CONF_USERNAME_LENGTH
	#define MQTT_CONF_USERNAME_LENGTH 13 // Recommended by MQTT Specification (12 + '\0')
#endif

#ifndef MQTT_CONF_PASSWORD_LENGTH
	#define MQTT_CONF_PASSWORD_LENGTH 13 // Recommended by MQTT Specification (12 + '\0')
#endif
```

This means the value can be upped to 20 and 30 by defining the flag in the compile phase.

ericmnel, 2016-01-04:
Are you suggesting that I put that somewhere in my project instead of changing Sming? (Makefile-user.mk maybe?)

etmmahi, 2016-01-05:
In the file Makefile-project.mk you could add these defines and then recompile the Sming library

```
CFLAGS += -DMQTT_CONF_USERNAME_LENGTH=20
CFLAGS += -DMQTT_CONF_PASSWORD_LENGTH=30
```

The other option is to increase the defaults in libemgtt.h as you described.
Disadvantage of increasing the defaults is the increased memory usage

ericmnel, 2016-01-05:
I don't see why there would be a memory difference changing it one place vs the other.   

Ive made the change in my local version of Sming, and I'll leave it up to the Sming development community to determine if the change would be useful enough for other people to put it into the main code.   12 characters seems pretty small for general use, and obviously IBM thinks so too.

raburton, 2016-01-05:
The difference in memory use isn't where it's done but if you need it done at all. The comment suggests this length is recommended by the spec so, assuming that's true then, them most people probably don't need it increasing. If you change the default in Sming and most people don't need it then their memory is being wasted, but if they do need it then it is easy enough to for them to modify the length with #defines.

ericmnel, 2016-01-05:
I looked through the spec and don't see any recommended limits:

http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349223

3.1.3.4 and 3.1.3.5 define the user name and password.   1.5.3 talks about strings in general, and there is a limit of 65535 bytes but no other limit mentioned that I could find (I searched for both '12' and '13' and found nothing).

raburton, 2016-01-05:
Well perhaps it came from an earlier version of the spec - I don't know, libmqtt is a third party component included in Sming, we didn't write it. The fact remains any extra length allocated to these strings that isn't used is a waste of memory on a device that has very little memory spare. While some people may choose to use names / passwords longer than 12 chars there is really no need to because 12 chars gives you a stupid number of combinations already. If people need to use the device with a service that insists on using longer strings then Sming can easily be made to do this, so there is no good reason to increase the default length.

closed 2016-01-05 by raburton