SmingHub/Sming #533: Timer class  is finished ?

flexiti, 2016-01-05:
@hreintke @anakod 
Can you look at this?  (Timer.cpp)

>	void Timer::setIntervalUs(uint64_t microseconds/* = 1000000*/)
{
....
....
	interval = microseconds / div;
		long_intvl_cntr = 0;
		long_intvl_cntr_lim = div;
	}
	else 
	{
		interval = microseconds;
		long_intvl_cntr = 0;
		long_intvl_cntr = 0;  
        }

Last line - It shouldn't be **long_intvl_cntr_lim** = 0 ?
And if support for this variable is finished  ?

danielnilsson9, 2016-01-06:
I think I wrote the code for long timer interval support, must have made a mistake. It should definitely be as you said:

long_intvl_cntr_lim = 0

Didn't show up in the test because you would have to reset the interval on a timer loaded with a large interval to a short interval to notice the bug.

flexiti, 2016-01-06:
@danielnilsson9 
Can You make PR please ?

commit a8023803 (2016-01-12): close #533
closed 2016-01-12 by raburton